ros-drivers/usb_cam #201: I rebuild the project in catkin workspace today, but there are something undefined in libusb_cam.so #200

Lyk77, 2022-11-02:
here are the error

/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to av_packet_from_data' /home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to sws_scale'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to av_frame_alloc' /home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to avcodec_close'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to av_frame_get_buffer' /home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to av_new_packet'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to avcodec_alloc_context3' /home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to av_log_set_level'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to avcodec_free_context' /home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to av_free'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to avcodec_open2' /home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to avcodec_send_packet'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to sws_getContext' /home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to av_image_get_buffer_size'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to av_parser_close' /home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to av_parser_init'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to avcodec_register_all' /home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to sws_freeContext'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to avcodec_receive_frame' /home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to av_image_copy_to_buffer'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_find_decoder'
collect2: error: ld returned 1 exit status
usb_cam/CMakeFiles/usb_cam_node.dir/build.make:129: recipe for target '/home/ljs/catkin_ws/devel/lib/usb_cam/usb_cam_node' failed
make[2]: *** [/home/ljs/catkin_ws/devel/lib/usb_cam/usb_cam_node] Error 1
CMakeFiles/Makefile2:4442: recipe for target 'usb_cam/CMakeFiles/usb_cam_node.dir/all' failed
make[1]: *** [usb_cam/CMakeFiles/usb_cam_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

Jeasonlee313, 2022-11-02:
I download the release version 0.3.6 and rebuild it in catkin workspace. The usb cam node work well, I this way can also work on your work space.

twdragon, 2022-11-02:
@Lyk77 it seemed that your installed versions of `libavutil` and `libavcodec` libraries are too old. Can you tell which versions do you have installed on your machine?

Lyk77, 2022-11-03:
> @Lyk77 it seemed that your installed versions of `libavutil` and `libavcodec` libraries are too old or the libraries are not linked properly. Can you tell which versions you have installed on your machine?

![4290127c3f3999bdc7dab4b0c7e66e995de990c4cd6b20c38fc59fc53b822bbaQzpcVXNlcnNcQWRtaW5pc3RyYXRvclxBcHBEYXRhXFJvYW1pbmdcRGluZ1RhbGtcMjM3NjU5NjYxNl92MlxJbWFnZUZpbGVzXDIyNDQwNDVcMTY2NzQ0MDQ2ODA3OF9VUkxfMjFkYzFjY2I0MDcxZTg5Y2M5ZTlhNzliMGYzNzRkNGY=](https://user-images.githubusercontent.com/109563954/199636756-dcd5b06a-2d1b-4445-8c9c-dd7d35a5f79d.png)

![e1767f921ae725084a2d20a90093dfef2cebc1d9eed36167e8920578a545f66cQzpcVXNlcnNcQWRtaW5pc3RyYXRvclxBcHBEYXRhXFJvYW1pbmdcRGluZ1RhbGtcMjM3NjU5NjYxNl92MlxJbWFnZUZpbGVzXDIyNDQwNDVcMTY2NzQ0MDQ1MzkyMF9VUkxfYWZiYzlkZjRhMGI0N2ZjNGFiNmYwZTVjODU0OTMwMTU=](https://user-images.githubusercontent.com/109563954/199636769-b5044978-a271-4fc0-9d64-ae6db8e75045.png)

Lyk77, 2022-11-03:
> @Lyk77 it seemed that your installed versions of `libavutil` and `libavcodec` libraries are too old or the libraries are not linked properly. Can you tell which versions you have installed on your machine?

As you said, it is because I did not link to the library. After I linked to the library, the compilation was successfull! I am sorry, I will end my question.

closed 2022-11-03 by Lyk77

ChadJing, 2022-11-04:
> 

How did you overcome the problem, can you be a little more detailed? Thanks a lot.

twdragon, 2022-11-04:
@ChadJing you need to rebuild the package with proper library locations set for the linker. Usually they are set using `*_LINK_LIBRARIES` variables in CMake, but in the worst case you can specify them via compiler flags (`-l` or `-L` for directories).

ChadJing, 2022-11-04:
@twdragon It  is still a little difficult for me, a new-bird, to understand, should I set them in the CMakeLists.txt whose path is under catkin_ws/src/usb_cam?