ros-drivers/usb_cam #202: libusb_cam.so: undefined reference

mistyk, 2022-11-14:
When i run catkin_make i've the following error, anyone have some advice for a newbie ? ;)

Thanks !

Log:

[100%] Linking CXX executable /home/jetson/catkin_ws/devel/lib/usb_cam/usb_cam_node
/usr/bin/ld: warning: libopencv_imgproc.so.3.2, needed by /home/jetson/catkin_ws/devel/lib/libusb_cam.so, may conflict with libopencv_imgproc.so.4.1
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_packet_from_data'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `sws_scale'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_frame_alloc'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_close'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_frame_get_buffer'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_new_packet'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_alloc_context3'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_log_set_level'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_free_context'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_free'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_open2'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_send_packet'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `sws_getContext'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_image_get_buffer_size'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_parser_close'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_parser_init'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_register_all'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `sws_freeContext'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_receive_frame'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_image_copy_to_buffer'
/home/jetson/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_find_decoder'
collect2: error: ld returned 1 exit status
usb_cam/CMakeFiles/usb_cam_node.dir/build.make:144: recipe for target '/home/jetson/catkin_ws/devel/lib/usb_cam/usb_cam_node' failed
make[2]: *** [/home/jetson/catkin_ws/devel/lib/usb_cam/usb_cam_node] Error 1
CMakeFiles/Makefile2:4437: recipe for target 'usb_cam/CMakeFiles/usb_cam_node.dir/all' failed
make[1]: *** [usb_cam/CMakeFiles/usb_cam_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2 -l2" failed

chenkefeng030, 2022-11-15:
> 
I reset to the 2020 version, and there is no problem.
hash code:3ce8ee1c47c03be37229bf5857ae9a309d8eb1e8

twdragon, 2022-11-15:
@chenkefeng030 you need to thoroughly check which version of FFMPEG/libavcodec you have. I see that some functions, `av_packet_from_data`, for example, are introduced in `libavutil` API from version 2.8 at least. It seems that your system is equipped with a very old version of `libavcodec` stack. Could you please send here the output of the command:
```
ffmpeg -version
```

mistyk, 2022-11-15:
@twdragon this is my ffmpeg version:
jetson@jetson-desktop:~$ ffmpeg -version
ffmpeg version 3.4.8-0ubuntu0.2 Copyright (c) 2000-2020 the FFmpeg developers
built with gcc 7 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04)
configuration: --prefix=/usr --extra-version=0ubuntu0.2 --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-librsvg --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
libavutil      55. 78.100 / 55. 78.100
libavcodec     57.107.100 / 57.107.100
libavformat    57. 83.100 / 57. 83.100
libavdevice    57. 10.100 / 57. 10.100
libavfilter     6.107.100 /  6.107.100
libavresample   3.  7.  0 /  3.  7.  0
libswscale      4.  8.100 /  4.  8.100
libswresample   2.  9.100 /  2.  9.100
libpostproc    54.  7.100 / 54.  7.100

mistyk, 2022-11-15:
> > 
> 
> I reset to the 2020 version, and there is no problem. hash code:3ce8ee1c47c03be37229bf5857ae9a309d8eb1e8

I've don't found this release, have you got links ?
Thanks!

chenkefeng030, 2022-11-16:
> > > 
> > 
> > 
> > I reset to the 2020 version, and there is no problem. hash code:3ce8ee1c47c03be37229bf5857ae9a309d8eb1e8
> 
> I've don't found this release, have you got links ? Thanks!

Maybe this is not the best solution，you can enter the following command under your local repository:
git reset --hard 3ce8ee1c47c03be37229bf5857ae9a309d8eb1e8.
I use jetson xavier nx and no problem.

twdragon, 2022-11-16:
@mistyk @chenkefeng030 I investigated the issue, it is a misspelled name of `libavcodec` library in the linker list of `CMakeLists.txt`. I will fix it very soon!

twdragon, 2022-11-16:
The issue reproduced for Melodic, investigation restarted.

mistyk, 2022-11-16:
@twdragon keep me informed ! Thanks you

twdragon, 2022-11-17:
@mistyk I examined `libavcodec` header files shipped with Ubuntu 18.04, 20.04 nd 22.04, so I will try to track and integrate the changes I discovered from there

mistyk, 2022-11-17:
@twdragon Thanks you ! i'm waiting for you to update my ROS ;) keep me in touch !

closed 2023-02-20 by twdragon

joker-wutong, 2024-05-15:
我也遇到了同样的问题：
`[ 75%] Linking CXX executable /home/agilex/idrive_ws/devel/lib/usb_cam/usb_cam_node
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `av_packet_from_data'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `sws_scale'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `av_frame_alloc'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_close'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `av_frame_get_buffer'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `av_new_packet'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_alloc_context3'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `av_log_set_level'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_free_context'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `av_free'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_open2'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_send_packet'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `sws_getContext'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `av_image_get_buffer_size'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `av_parser_close'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `av_parser_init'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_register_all'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `sws_freeContext'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_receive_frame'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `av_image_copy_to_buffer'
/home/agilex/idrive_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_find_decoder'
`
 i'm on Jetson nano with melodic, this is my version with latest ffmpeg for bionic

`ffmpeg version 3.4.11-0ubuntu0.1 Copyright (c) 2000-2022 the FFmpeg developers
built with gcc 7 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04)
configuration: --prefix=/usr --extra-version=0ubuntu0.1 --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-librsvg --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
libavutil      55. 78.100 / 55. 78.100
libavcodec     57.107.100 / 57.107.100
libavformat    57. 83.100 / 57. 83.100
libavdevice    57. 10.100 / 57. 10.100
libavfilter     6.107.100 /  6.107.100
libavresample   3.  7.  0 /  3.  7.  0
libswscale      4.  8.100 /  4.  8.100
libswresample   2.  9.100 /  2.  9.100
libpostproc    54.  7.100 / 54.  7.100
`

怎么解决啊，各位全栈工程师们，技术大佬啊，求教！
help！！！

commit 650d6ef8 (2022-11-16): Fixed library names for pkg-config

- Fixed library names in the `pkg-config` section of CMakeLists.txt
- Closes #202