ros-drivers/usb_cam #205: libusb_cam.so: undefined reference "make -j4 -l4" failed

04633435, 2022-11-18:
Hi there, I was running `$ catkin_make` but I got an error very similar to #202, the only difference is the last line, "make -j4 -l4" in my case, "make -j2 -l2" in #202.

My CMakeLists.txt is up to date. Is there any advice to fix this error? Thanks~

The whole working log is attached below: 

```
$ catkin_make
Base path: /home/olivia/camera_ws
Source space: /home/olivia/camera_ws/src
Build space: /home/olivia/camera_ws/build
Devel space: /home/olivia/camera_ws/devel
Install space: /home/olivia/camera_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/olivia/camera_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/olivia/camera_ws/build"
####
[ 50%] Built target usb_cam
[ 75%] Linking CXX executable /home/olivia/camera_ws/devel/lib/usb_cam/usb_cam_node
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `av_packet_from_data'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `sws_scale'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `av_frame_alloc'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_close'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `av_frame_get_buffer'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `av_new_packet'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_alloc_context3'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `av_log_set_level'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_free_context'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `av_free'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_open2'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_send_packet'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `sws_getContext'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `av_image_get_buffer_size'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `av_parser_close'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `av_parser_init'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_register_all'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `sws_freeContext'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_receive_frame'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `av_image_copy_to_buffer'
/home/olivia/camera_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_find_decoder'
collect2: error: ld returned 1 exit status
usb_cam/CMakeFiles/usb_cam_node.dir/build.make:170: recipe for target '/home/olivia/camera_ws/devel/lib/usb_cam/usb_cam_node' failed
make[2]: *** [/home/olivia/camera_ws/devel/lib/usb_cam/usb_cam_node] Error 1
CMakeFiles/Makefile2:1326: recipe for target 'usb_cam/CMakeFiles/usb_cam_node.dir/all' failed
make[1]: *** [usb_cam/CMakeFiles/usb_cam_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

twdragon, 2022-11-18:
@04633435 it is just a build pipeline thread count, so I will close in favour of #202 until the solution will be implemented (it is possible, I just need a bit of time).

closed 2022-11-18 by twdragon

04633435, 2022-11-18:
> 

Thanks for reply!

Got it~ Just one more question

Is it possible to workaround this error by using the old version?

mistyk, 2022-11-18:
@04633435 have you got a link of a working version ?

twdragon, 2022-11-18:
@mistyk try to use a version from Release: https://github.com/ros-drivers/usb_cam/releases/tag/0.4.0

mistyk, 2022-11-18:
@twdragon i'm on ROS1 not 2 i can't use this package

twdragon, 2022-11-18:
@mistyk try to checkout the commit 6311f2667472728cded862a4a09d9097dcf06472

04633435, 2022-11-21:
@twdragon Thanks for the help. The old version works.
@mistyk We are on ROS1 as well, [0.3.6](https://github.com/ros-drivers/usb_cam/releases/tag/0.3.6) works for us,  build without error. Same error msg as you using 0.4.0