ros-drivers/usb_cam #211: Memory leak in develop branch, in  UsbCam::mjpeg2rgb()

AIRCAP, 2022-12-09:
https://github.com/ros-drivers/usb_cam/blob/develop/src/usb_cam.cpp line# 501

Calling both `av_new_packet() `and `av_packet_from_data()` allocates the buffer within `avpkt.buf` twice, causing usb_cam to run out of memory.

Possible solution: call `av_init_packet(&avpkt)` instead of `av_new_packet(&avpkt, len)`  which does not initialize this buffer, then call `av_packet_from_data()` which does.

Not making a pull request, since this is in process of refactoring anyway.

twdragon, 2022-12-09:
@AIRCAP thank you for pointing this! Will be considered in refactoring

flynneva, 2022-12-12:
@twdragon this seems to only be an issue on the ROS 1 `develop` branch...since I think the calls referenced here have been replaced with something else.  Might be something that @AIRCAP could confirm though...

twdragon, 2022-12-12:
@flynneva you are right, this is only a ROS1 issue and will be refactored. I hope to start active rewriting before Christmas

PR #233 (2023-03-02): fix memory leak in usb_cam.cpp
we should do step by step, not making a big PR.
also this is not a refactoring, but a issue solution.
the priority is higher than refactoring.

with this PR, the memory leaks disappear.
closes #211 

https://user-images.githubusercontent.com/9300063/222413841-5e0ac1d6-9818-40d2-8089-7c96ffd85378.mp4
PR #231 (2023-04-07): [refactored version of #221] Total refactoring
- The node implemented as singleton with monopolistic device access model (one node per device)
- Moved and separated into different blocks the the code using  V4L2, ROS and FFMPEG APIs
- Removed message pointers completely, the node uses runtime-constant shared object addresses
- Debugged FFMPEG decoder algorithm, removed memory leak. Closes #211
- Added (using @flynneva's code) the conditional compilation directives to exploit properly versioned libavcodec API. Closes #207
- Closes #203
- Added separated launch files to run the node and to test  [... cut]
commit a6635621 (2023-04-07): Total refactoring
	- The node implemented as singleton with monopolistic device
	  access model (one node per device)
	- Moved and separated into different blocks the the code using
	  V4L2, ROS and FFMPEG APIs
	- Removed message pointers completely, the node uses runtime-
	  constant shared object addresses
	- Debugged FFMPEG decoder algorithm, removed memory leak.
	  Closes #211
	- Added (using @flynneva's code) the conditional compilation
	  directives to exploit properly versioned libavcodec API.
	  Closes #207
	- Closes #203
	- Added separated launch files to run the node and to test it
	 [... cut]
closed 2023-03-02 by twdragon