ros-drivers/usb_cam #213: camera select

g2181053, 2022-12-13:
I want to use a usb camera with rviz2.

The HD cam on the PC is selected and the camera connected via usb is not.

It seems that the HD cam on the PC is video0/video1 and the camera I want to see is video2/video3.

The cameras I want to see are
https://www.theimagingsource.com/products/industrial-cameras/usb-3.0-color/dfk27bur0135/

flynneva, 2022-12-15:
@g2181053 so ROS 2 it is recommended (and tested) to use colcon build to build the usb_cam (and most other ROS 2) packages.

You shouldn't have to rebuild if you are just adjusting the params file. Just pass it in via CLI and you should be good.

g2181053, 2022-12-18:
Thanks for the reply.

I have colon_built the usb_cam since then. It built without any problems. However, there was no config file, so I then downloaded and pasted the config file into the usb_cam file.

(The hierarchy is as follows
~catkin_ws/src/usb_cam/config)

And that the desired camera is /dev/video0".
$ v4l2-ctl --list-devices
to check that the desired camera is in "/dev/video0".

I then executed the following commands

$ source /opt/ros/galactic/setup.bash
$ ros2 run usb_cam usb_cam_node_exe --ros-args --params-file ~/catkin_ws/src/usb_cam/config/params.yaml

The results were as follows

[INFO] [1671371038.193622567] [usb_cam]: camera_name value: test_camera
[WARN] [1671371038.193695301] [usb_cam]: framerate: 60.000000
[INFO] [1671371038.202750261] [usb_cam]: camera calibration URL: package://usb_cam/config/camera_info.yaml
[INFO] [1671371038.204199038] [usb_cam]: Starting 'test_camera' (/dev/video0) at 1280x960 via mmap (yuyv) at 60 FPS
[INFO] [1671371038.299294035] [usb_cam]: Capability flag: 0x1000
[INFO] [1671371038.300109749] [usb_cam]: Set framerate to be 60
[INFO] [1671371038.300855453] [usb_cam]: This Cameras Supported Formats:
[INFO] [1671371038.300913692] [usb_cam]: 8-bit Bayer GRGR/BGBG[Index: 0, Type: 1, Flags: 0, PixelFormat: 47425247]
[INFO] [1671371038.300939293] [usb_cam]: width: 1280 x height: 960
[INFO] [1671371038.300967684] [usb_cam]: 1 1 / 60
[INFO] [1671371038.300998350] [usb_cam]: 1 1 / 50
[INFO] [1671371038.301005841] [usb_cam]: 1 1 / 40

Then the command execution ended and I could not see any image_raw in rviz2.
As far as I can see from the results, the parameter file in the config file is readable, so was there something wrong with the way the parameter file was filled out?

I filled out the params.yaml as follows

/**:
    ros__parameters:
      video_device: "/dev/video0"
      framerate: 60.0
      io_method: "mmap"
      frame_id: "camera"
      pixel_format: "yuyv"
      color_format: "yuv422p"
      image_width: 1280
      image_height: 960
      camera_name: "test_camera"
      camera_info_url: "package://usb_cam/config/camera_info.yaml"

      I did not change io_method, frame_id, pixel_format, and color_format because I was not sure about them.

What do you think?

Translated with www.DeepL.com/Translator (free version)

g2181053, 2022-12-18:
Thanks for the reply.

I have colon_built the usb_cam since then. It built without any problems. However, there was no config file, so I then downloaded and pasted the config file into the usb_cam file.

(The hierarchy is as follows
~catkin_ws/src/usb_cam/config)

And that the desired camera is /dev/video0".
$ v4l2-ctl --list-devices
to check that the desired camera is in "/dev/video0".

I then executed the following commands

$ source /opt/ros/galactic/setup.bash
$ ros2 run usb_cam usb_cam_node_exe --ros-args --params-file ~/catkin_ws/src/usb_cam/config/params.yaml

The results were as follows

[INFO] [1671371038.193622567] [usb_cam]: camera_name value: test_camera
[WARN] [1671371038.193695301] [usb_cam]: framerate: 60.000000
[INFO] [1671371038.202750261] [usb_cam]: camera calibration URL: package://usb_cam/config/camera_info.yaml
[INFO] [1671371038.204199038] [usb_cam]: Starting 'test_camera' (/dev/video0) at 1280x960 via mmap (yuyv) at 60 FPS
[INFO] [1671371038.299294035] [usb_cam]: Capability flag: 0x1000
[INFO] [1671371038.300109749] [usb_cam]: Set framerate to be 60
[INFO] [1671371038.300855453] [usb_cam]: This Cameras Supported Formats:
[INFO] [1671371038.300913692] [usb_cam]: 8-bit Bayer GRGR/BGBG[Index: 0, Type: 1, Flags: 0, PixelFormat: 47425247]
[INFO] [1671371038.300939293] [usb_cam]: width: 1280 x height: 960
[INFO] [1671371038.300967684] [usb_cam]: 1 1 / 60
[INFO] [1671371038.300998350] [usb_cam]: 1 1 / 50
[INFO] [1671371038.301005841] [usb_cam]: 1 1 / 40

Omitted

INFO] [1671371216.124534449] [usb_cam]: width: 640 x height: 480
[INFO] [1671371216.124541653] [usb_cam]: 1 1 / 120
[INFO] [1671371216.124548754] [usb_cam]: 1 1 / 90
[INFO] [1671371216.124554974] [usb_cam]: 1 1 / 60
[INFO] [1671371216.124561067] [usb_cam]: 1 1 / 30
[INFO] [1671371216.124569159] [usb_cam]: 1 1 / 15
[INFO] [1671371216.124575866] [usb_cam]: 1 1 / 10
[INFO] [1671371216.124615288] [usb_cam]: starting timer 16

Then the command execution ended and I could not see any image_raw in rviz2.
As far as I can see from the results, the parameter file in the config file is readable, so was there something wrong with the way the parameter file was filled out?

I filled out the params.yaml as follows

/**:
    ros__parameters:
      video_device: "/dev/video0"
      framerate: 60.0
      io_method: "mmap"
      frame_id: "camera"
      pixel_format: "yuyv"
      color_format: "yuv422p"
      image_width: 1280
      image_height: 960
      camera_name: "test_camera"
      camera_info_url: "package://usb_cam/config/camera_info.yaml"

      I did not change io_method, frame_id, pixel_format, and color_format because I was not sure about them.

What do you think?

flynneva, 2022-12-18:
@g2181053 the console output helps a lot! Thanks!

It looks like your device only supports 8-bit Bayer (listed in the output).

You will most likely have to select the proper pixel and color format then I think.

I'm not by my computer to look and I can't remember the right string to use for that option...but that should be at least where you start to look I think.

flynneva, 2022-12-19:
@g2181053 so I think you are confusing things here.

First, you have formers supported by your physical device. These are the formats that your physical hardware can output. In your case your camera supports grgb and a few others.

Next you have formats that your software (aka usb_cam in this case) supports. This driver supports a few different formats, you can see them listed here in the function on line 156:

https://github.com/ros-drivers/usb_cam/blob/ros2/include/usb_cam/utils.hpp

I think if I were you I'd try the mono10 and the rgb24 format strings. I think those might be the ones that work with your camera. If not, then we might need to add another format to usb_cam for grbg (not sure the specs for your format though, you'd have to provide them)

g2181053, 2022-12-21:
Thanks for the reply.

I immediately changed the pixel_format to rgb24, checked it with rviz2 and the image came up!!!

However, the output screen was divided into 9 parts and the same image was shown on each part of the screen. Also, the image is not stable.

Is there anything you can tell us about this?

flynneva, 2022-12-21:
@g2181053 hmm that sounds like it's not the right format then...or there might be something wrong with the conversion method we have in this repo usb_cam.

Could you take a screen shot and share the console output again?

For now maybe try mono8 or mono10 formats instead - those will be in black and white though unfortunately I think.

g2181053, 2022-12-22:
Below are the results of the program run.

$ ros2 run usb_cam usb_cam_node_exe --ros-args --params-file ~/catkin_ws/src/usb_cam/config/params.yaml

[INFO] [1671699395.341552486] [usb_cam]: camera_name value: test_camera
[WARN] [1671699395.341904599] [usb_cam]: framerate: 30.000000
[INFO] [1671699395.342839224] [usb_cam]: camera calibration URL: package://usb_cam/config/camera_info.yaml
[INFO] [1671699395.343718959] [usb_cam]: Starting 'test_camera' (/dev/video0) at 640x480 via mmap (rgb24) at 30 FPS
[INFO] [1671699395.440508389] [usb_cam]: Capability flag: 0x1000
[INFO] [1671699395.441207550] [usb_cam]: Set framerate to be 30
[INFO] [1671699395.441605256] [usb_cam]: This Cameras Supported Formats:
[INFO] [1671699395.441661723] [usb_cam]:   8-bit Bayer GRGR/BGBG[Index: 0, Type: 1, Flags: 0, PixelFormat: 47425247]
[INFO] [1671699395.441693082] [usb_cam]:   width: 1280 x height: 960
[INFO] [1671699395.441702701] [usb_cam]:   1 1 / 60
[INFO] [1671699395.441723468] [usb_cam]:   1 1 / 50
[INFO] [1671699395.441729669] [usb_cam]:   1 1 / 40
[INFO] [1671699395.441735485] [usb_cam]:   1 1 / 30
[INFO] [1671699395.441741256] [usb_cam]:   1 1 / 20
[INFO] [1671699395.441774898] [usb_cam]:   1 1 / 10
[INFO] [1671699395.441803309] [usb_cam]:   width: 1280 x height: 720
[INFO] [1671699395.441810494] [usb_cam]:   1 1 / 80
[INFO] [1671699395.441831397] [usb_cam]:   1 1 / 60
[INFO] [1671699395.441857710] [usb_cam]:   1 1 / 50
[INFO] [1671699395.441864053] [usb_cam]:   1 1 / 40
[INFO] [1671699395.441870674] [usb_cam]:   1 1 / 30
[INFO] [1671699395.441877490] [usb_cam]:   1 1 / 20
[INFO] [1671699395.441883983] [usb_cam]:   1 1 / 10
[INFO] [1671699395.441890539] [usb_cam]:   width: 1024 x height: 768
[INFO] [1671699395.441897063] [usb_cam]:   1 1 / 80
[INFO] [1671699395.441919716] [usb_cam]:   1 1 / 70
[INFO] [1671699395.441946361] [usb_cam]:   1 1 / 60
[INFO] [1671699395.441952640] [usb_cam]:   1 1 / 40
[INFO] [1671699395.441959099] [usb_cam]:   1 1 / 30
[INFO] [1671699395.441965873] [usb_cam]:   1 1 / 20
[INFO] [1671699395.441972503] [usb_cam]:   1 1 / 10
[INFO] [1671699395.441978966] [usb_cam]:   width: 640 x height: 480
[INFO] [1671699395.441985530] [usb_cam]:   1 1 / 120
[INFO] [1671699395.442005905] [usb_cam]:   1 1 / 90
[INFO] [1671699395.442032615] [usb_cam]:   1 1 / 60
[INFO] [1671699395.442043450] [usb_cam]:   1 1 / 30
[INFO] [1671699395.442055358] [usb_cam]:   1 1 / 15
[INFO] [1671699395.442067584] [usb_cam]:   1 1 / 10
[INFO] [1671699395.442076708] [usb_cam]:   8-bit Greyscale[Index: 1, Type: 1, Flags: 0, PixelFormat: 59455247]
[INFO] [1671699395.442084523] [usb_cam]:   width: 1280 x height: 960
[INFO] [1671699395.442091608] [usb_cam]:   1 1 / 60
[INFO] [1671699395.442098252] [usb_cam]:   1 1 / 50
[INFO] [1671699395.442105002] [usb_cam]:   1 1 / 40
[INFO] [1671699395.442111744] [usb_cam]:   1 1 / 30
[INFO] [1671699395.442118421] [usb_cam]:   1 1 / 20
[INFO] [1671699395.442125052] [usb_cam]:   1 1 / 10
[INFO] [1671699395.442131998] [usb_cam]:   width: 1280 x height: 720
[INFO] [1671699395.442138900] [usb_cam]:   1 1 / 80
[INFO] [1671699395.442145457] [usb_cam]:   1 1 / 60
[INFO] [1671699395.442152122] [usb_cam]:   1 1 / 50
[INFO] [1671699395.442158732] [usb_cam]:   1 1 / 40
[INFO] [1671699395.442168571] [usb_cam]:   1 1 / 30
[INFO] [1671699395.442181427] [usb_cam]:   1 1 / 20
[INFO] [1671699395.442189493] [usb_cam]:   1 1 / 10
[INFO] [1671699395.442196607] [usb_cam]:   width: 1024 x height: 768
[INFO] [1671699395.442203781] [usb_cam]:   1 1 / 80
[INFO] [1671699395.442210461] [usb_cam]:   1 1 / 70
[INFO] [1671699395.442217427] [usb_cam]:   1 1 / 60
[INFO] [1671699395.442225879] [usb_cam]:   1 1 / 40
[INFO] [1671699395.442238062] [usb_cam]:   1 1 / 30
[INFO] [1671699395.442246097] [usb_cam]:   1 1 / 20
[INFO] [1671699395.442252810] [usb_cam]:   1 1 / 10
[INFO] [1671699395.442259993] [usb_cam]:   width: 640 x height: 480
[INFO] [1671699395.442267066] [usb_cam]:   1 1 / 120
[INFO] [1671699395.442273860] [usb_cam]:   1 1 / 90
[INFO] [1671699395.442281411] [usb_cam]:   1 1 / 60
[INFO] [1671699395.442288214] [usb_cam]:   1 1 / 30
[INFO] [1671699395.442295011] [usb_cam]:   1 1 / 15
[INFO] [1671699395.442301535] [usb_cam]:   1 1 / 10
[INFO] [1671699395.442309208] [usb_cam]:   16-bit Greyscale[Index: 2, Type: 1, Flags: 0, PixelFormat: 20363159]
[INFO] [1671699395.442334076] [usb_cam]:   width: 1280 x height: 960
[INFO] [1671699395.442344947] [usb_cam]:   1 1 / 70
[INFO] [1671699395.442356797] [usb_cam]:   1 1 / 60
[INFO] [1671699395.442367549] [usb_cam]:   1 1 / 50
[INFO] [1671699395.442374828] [usb_cam]:   1 1 / 40
[INFO] [1671699395.442381568] [usb_cam]:   1 1 / 30
[INFO] [1671699395.442388207] [usb_cam]:   1 1 / 20
[INFO] [1671699395.442394989] [usb_cam]:   1 1 / 10
[INFO] [1671699395.442405722] [usb_cam]:   width: 640 x height: 480
[INFO] [1671699395.442417578] [usb_cam]:   1 1 / 120
[INFO] [1671699395.442424917] [usb_cam]:   1 1 / 90
[INFO] [1671699395.442431618] [usb_cam]:   1 1 / 60
[INFO] [1671699395.442438506] [usb_cam]:   1 1 / 30
[INFO] [1671699395.442445198] [usb_cam]:   1 1 / 15
[INFO] [1671699395.442451869] [usb_cam]:   1 1 / 10
[INFO] [1671699395.442489811] [usb_cam]: starting timer 33

I have just contacted the person in charge of the camera.
The formats supported by this camera were as follows

Y800, Y16, RGB32, RGB64

Since there is no RGB24, is it impossible to project images correctly here?

flynneva, 2023-04-02:
@g2181053 I've just re-written the way we add new supported formats here.  Feel free to take a look and see if you can use any of the supported formats.

If not, someone will need to add a new pixel format class for whatever format you want to support.

flynneva, 2023-04-09:
Closing since I believe this is resolved

closed 2023-04-09 by flynneva