ros-drivers/usb_cam #214: [ROS2] symbol lookup error when starting usb_cam_node_exe

greinet, 2022-12-14:
Hey,

I get the following error when running the executable:

```
$ ros2 run usb_cam usb_cam_node_exe 
[INFO] [1671021050.892231053] [usb_cam]: camera_name value: default_cam
[WARN] [1671021050.892404783] [usb_cam]: framerate: 30.000000
[INFO] [1671021050.895138645] [usb_cam]: using default calibration URL
[INFO] [1671021050.895189558] [usb_cam]: camera calibration URL: file:///home/uam/.ros/camera_info/default_cam.yaml
[ERROR] [1671021050.895283575] [camera_calibration_parsers]: Unable to open camera calibration file [/home/uam/.ros/camera_info/default_cam.yaml]
[WARN] [1671021050.895306711] [usb_cam]: Camera calibration file /home/uam/.ros/camera_info/default_cam.yaml not found
[INFO] [1671021050.895375512] [usb_cam]: Starting 'default_cam' (/dev/video0) at 640x480 via mmap (yuyv, yuv422p) at 30 FPS
/home/uam/Downloads/temp/cam_newest/install/usb_cam/lib/usb_cam/usb_cam_node_exe: symbol lookup error: /home/uam/Downloads/temp/cam_newest/install/usb_cam/lib/libusb_cam_node.so: undefined symbol: _ZN7usb_cam6UsbCam5startERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_5utils9io_methodENS9_12pixel_formatENS9_12color_formatEjji
```
I am building the current `ros2` branch locally with ROS2 Foxy on a NVIDIA Orin (aarch64/arm64 architecture).  

As I used this library before, I tried going back in commits to find where the issue first started. [This commit](https://github.com/ros-drivers/usb_cam/commit/027b5dedb58344687cd3fae7052bf71e7958476b) was the first one that started to fail with a similar error:

```
$ ros2 run usb_cam usb_cam_node_exe --ros-args --params-file params.yaml
[INFO] [1671021461.208066691] [usb_cam]: camera_name value: test_camera
[WARN] [1671021461.208262182] [usb_cam]: framerate: 30.000000
[INFO] [1671021461.211464913] [usb_cam]: camera calibration URL: package://usb_cam/config/camera_info.yaml
[INFO] [1671021461.212513727] [usb_cam]: Starting 'test_camera' (/dev/video0) at 1920x1080 via mmap (h264, yuv420p) at 30 FPS
/home/uam/Downloads/temp/cam_ws/install/usb_cam/lib/usb_cam/usb_cam_node_exe: symbol lookup error: /home/uam/Downloads/temp/cam_ws/install/usb_cam/lib/libusb_cam.so: undefined symbol: _Z14av_image_allocPPhPiii13AVPixelFormati
```
My understanding is, that it does not recognize the method `av_image_alloc` in [line 111 of that commit](https://github.com/ros-drivers/usb_cam/commit/027b5dedb58344687cd3fae7052bf71e7958476b#diff-e24935df8b248d194d83458dd57a9606f021654168887659350b2e563a92f6a4R111).  

I also listed my installed libavcodec version, resulting in 58 as far as i can tell, which should provide the wanted method i guess:
```
$ sudo apt list --installed | grep -i libavcodec
libavcodec-dev/focal-updates,focal-security,now 7:4.2.7-0ubuntu0.1 arm64 [installed,automatic]
libavcodec58/focal-updates,focal-security,now 7:4.2.7-0ubuntu0.1 arm64 [installed]
```
I guess this might be related to https://github.com/ros-drivers/usb_cam/issues/203. What libavcodec versions are currently supported and what can i do to get this package running on my setup?

flynneva, 2023-01-14:
@greinet apologies about the inconvenience!  You are right the error was related / caused by the work being done for #203, and should be fixed now after #212 was just merged.  Could you test out that branch and let me know if you are still having issues?

greinet, 2023-01-18:
Hey @flynneva, thanks for the response. I tested the new version and indeed no more error regarding libavcodec occurs. However, I can now only get my camera working with the pixel format yuyv and not with mjpeg or h264. My camera supports yuyv and mjpeg on /dev/video0:
```
uam@ubuntu:~/ros_ws_usb_cam$ sudo v4l2-ctl --list-formats -d /dev/video0
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
	[1]: 'YUYV' (YUYV 4:2:2)
```
and H.264 on /dev/video2:
```
uam@ubuntu:~/ros_ws_usb_cam$ sudo v4l2-ctl --list-formats -d /dev/video2
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'H264' (H.264, compressed)
```
When I start the usb_cam_node_exe with default settings, it creates the yuyv stream on /dev/video0 just fine and it also display the possible pixel formats:
```
uam@ubuntu:~/ros_ws_usb_cam$ ros2 run usb_cam usb_cam_node_exe 
[INFO] [1674030360.165441606] [usb_cam]: camera_name value: default_cam
[WARN] [1674030360.165624776] [usb_cam]: framerate: 30.000000
[INFO] [1674030360.167132254] [usb_cam]: using default calibration URL
[INFO] [1674030360.167181502] [usb_cam]: camera calibration URL: file:///home/uam/.ros/camera_info/default_cam.yaml
[ERROR] [1674030360.167279328] [camera_calibration_parsers]: Unable to open camera calibration file [/home/uam/.ros/camera_info/default_cam.yaml]
[WARN] [1674030360.167300832] [usb_cam]: Camera calibration file /home/uam/.ros/camera_info/default_cam.yaml not found
[INFO] [1674030360.167364033] [usb_cam]: Starting 'default_cam' (/dev/video0) at 640x480 via mmap (yuyv, yuv422p) at 30 FPS
[INFO] [1674030360.783876377] [usb_cam]: This devices supproted formats:
[INFO] [1674030360.784012699] [usb_cam]: 	Motion-JPEG: 1920 x 1080 (30 Hz)
...
[INFO] [1674030360.785169932] [usb_cam]: 	YUYV 4:2:2: 1920 x 1080 (5 Hz)
...
[INFO] [1674030360.785962103] [usb_cam]: Timer triggering every 33 ms
```
However, when i switch to using pixel format mjpeg and color format yuv420p the node just closes without an error:
```
uam@ubuntu:~/ros_ws_usb_cam$ ros2 run usb_cam usb_cam_node_exe --ros-args -p pixel_format:=mjpeg -p color_format:=yuv420p
[INFO] [1674030334.504697822] [usb_cam]: camera_name value: default_cam
[WARN] [1674030334.504870145] [usb_cam]: framerate: 30.000000
[INFO] [1674030334.506204757] [usb_cam]: using default calibration URL
[INFO] [1674030334.506254806] [usb_cam]: camera calibration URL: file:///home/uam/.ros/camera_info/default_cam.yaml
[ERROR] [1674030334.506351927] [camera_calibration_parsers]: Unable to open camera calibration file [/home/uam/.ros/camera_info/default_cam.yaml]
[WARN] [1674030334.506374743] [usb_cam]: Camera calibration file /home/uam/.ros/camera_info/default_cam.yaml not found
[INFO] [1674030334.506437624] [usb_cam]: Starting 'default_cam' (/dev/video0) at 640x480 via mmap (mjpeg, yuv420p) at 30 FPS
uam@ubuntu:~/ros_ws_usb_cam$
```
The same happens when i try to use H.264 on /dev/video2:
```
uam@ubuntu:~/ros_ws_usb_cam$ ros2 run usb_cam usb_cam_node_exe --ros-args -p video_device:=/dev/video2 -p pixel_format:=h264
[INFO] [1674030495.238309143] [usb_cam]: camera_name value: default_cam
[WARN] [1674030495.238604858] [usb_cam]: framerate: 30.000000
[INFO] [1674030495.240066858] [usb_cam]: using default calibration URL
[INFO] [1674030495.240183499] [usb_cam]: camera calibration URL: file:///home/uam/.ros/camera_info/default_cam.yaml
[ERROR] [1674030495.240340077] [camera_calibration_parsers]: Unable to open camera calibration file [/home/uam/.ros/camera_info/default_cam.yaml]
[WARN] [1674030495.240363853] [usb_cam]: Camera calibration file /home/uam/.ros/camera_info/default_cam.yaml not found
[INFO] [1674030495.240431694] [usb_cam]: Starting 'default_cam' (/dev/video2) at 640x480 via mmap (h264, yuv422p) at 30 FPS
uam@ubuntu:~/ros_ws_usb_cam$
```
Is there an issue with my configuration or my device?

flynneva, 2023-03-09:
@greinet you might have to test out different color formats that are compatible with the pixel format you chose. I am not sure what exactly your error is and it is hard for me to debug remotely.

flynneva, 2023-04-02:
@greinet fyi MJPEG is now supported on the `ros2` branch. just use the latest `ros2` branch and specify `mjpeg2rgb` in the pixel format parameter in the config file.

👍🏼

closed 2023-04-02 by flynneva