ros-drivers/usb_cam #216: How to build usb_cam on ROS2 rolling installed in docker image using Dockerfile?

fghoussen, 2023-01-14:
How to build `usb_cam` on ROS2 rolling installed in docker image using Dockerfile?

I have installed ROS2 rolling, following the official ROS2 doc, in a docker image using a Dockerfile: from a starting image `ubuntu` (jammy), I created an image `ubuntu/ros2`. On `ubuntu/ros2`, I can source the local bash setup script and run the hello world (listener - talker example runs OK).

Now, I need to fetch USB camera images. From  `ubuntu/ros2`, I'd like to create another image  `ubuntu/ros2drv` so I followed the same process:

- Writing a new Dockerfile which looks like:

```
$ cat Dockerfile 
FROM ubuntu/ros2
RUN echo 'source /home/ros2_rolling/install/local_setup.bash' >> ~/.bashrc
COPY install_usb_cam.sh /tmp
RUN cd /tmp && bash -x install_usb_cam.sh

$ cat install_usb_cam.sh 
#!/bin/bash
mkdir -p /home/ros_workspaces
cd /home/ros_workspaces
git clone https://github.com/ros-drivers/usb_cam
cd usb_cam
rosdep update
colcon build --symlink-install

``` 

Note that `/home/ros2_rolling/install/local_setup.bash` is sourced (`rosdep update` runs OK + catkin_make should be found), but, when building a new docker image I get

```
Add distro "rolling"
updated cache in /root/.ros/rosdep/sources.cache
+ colcon build --symlink-install
Starting >>> usb_cam
--- stderr: usb_cam
CMake Error at CMakeLists.txt:7 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< usb_cam [2.29s, exited with code 1]

Summary: 0 packages finished [2.85s]
  1 package failed: usb_cam
  1 package had stderr output: usb_cam
The command '/bin/sh -c cd /tmp && bash -x install_usb_cam.sh' returned a non-zero code: 1

```

How to fix this?

flynneva, 2023-01-14:
@fghoussen it looks like you are trying to build for ROS 2 rolling but I don't see you checking out the `ros2` branch in your dockerfile.

Could you try to add a `git checkout ros2` and see if it works?

flynneva, 2023-01-14:
@fghoussen also just  tip: if you are trying to use this package inside a docker container make sure you start the container with `--privileged` so to expose the `/dev/video*` devices to your docker env.

fghoussen, 2023-01-14:
@flynneva: Unfortunately not working?!... Any clue?

> Could you try to add a git checkout ros2 and see if it works?

Didn't know there was a `ros2` branch! :) Unfortunately doesn't work:

```
+ git clone https://github.com/ros-drivers/usb_cam
Cloning into 'usb_cam'...
+ cd usb_cam
+ git checkout ros2
Switched to a new branch 'ros2'
Branch 'ros2' set up to track remote branch 'ros2' from 'origin'.
...
updated cache in /root/.ros/rosdep/sources.cache
+ colcon build --symlink-install
Starting >>> usb_cam
--- stderr: usb_cam
CMake Error at CMakeLists.txt:13 (find_package):
  By not providing "Findament_cmake_auto.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ament_cmake_auto", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake_auto"
  with any of the following names:

    ament_cmake_autoConfig.cmake
    ament_cmake_auto-config.cmake

  Add the installation prefix of "ament_cmake_auto" to CMAKE_PREFIX_PATH or
  set "ament_cmake_auto_DIR" to a directory containing one of the above
  files.  If "ament_cmake_auto" provides a separate development package or
  SDK, be sure it has been installed.
```

> make sure you start the container with --privileged so to expose the /dev/video* devices to your docker env.

Thanks for the tip! Hope to use it when building is OK...

flynneva, 2023-01-14:
@fghoussen usually that error means you did not `source /opt/ros/rolling/setup.bash` before trying to build. Could you double check you have sourced the underlay properly?

fghoussen, 2023-01-14:
Weird: got compilation success running same commands lines by hand in docker container. I could commit an image but no way to automate the process using Dockerfile?!... 

This is supposed to do the job `RUN echo 'source /home/ros2_rolling/install/local_setup.bash' >> ~/.bashrc` but seems it doesn't. Modyfing the initial Dockerfile this way `RUN cd /tmp && source /home/ros2_rolling/install/local_setup.bash && bash -x install_usb_cam.sh` doesn't help and fires ` /bin/sh: 1: source: not found ` in docker container

flynneva, 2023-01-14:
@fghoussen when doing RUN commands in docker it's like opening a new terminal.

To get it to work you need to combine the source and compile commands into one RUN step in your docker file by using a combo of ; and && to run multiple commands.

fghoussen, 2023-01-14:
Finally got it to work with `ENTRYPOINT [".", "/home/ros2_rolling/install/local_setup.bash"]` in Dockerfile, and, `. /home/ros2_rolling/install/local_setup.bash` in the sh script. Not sure why previous attempts failed... Anyway.

closed 2023-01-14 by fghoussen