ros-drivers/usb_cam #219: How to run usb_cam on ROS2 rolling installed in docker image using Dockerfile?

fghoussen, 2023-01-15:
How to run usb_cam on ROS2 rolling installed in docker image using Dockerfile?

After #216 (where I suppressed `ENTRYPOINT` to source `local_setup.bash` to prevent error that broke `docker run` launch), I get:

```
$ sudo docker run -it --rm --network host --privileged -e DISPLAY=:0 -v /tmp/.X11-unix:/tmp/.X11-unix ubuntu/ros2drv /bin/bash
root@raspberrypi:/home# . /home/ros2_rolling/install/local_setup.bash
root@raspberrypi:/home# . /home/ros_workspaces/usb_cam/install/local_setup.bash
root@raspberrypi:/home# roscore
bash: roscore: command not found
root@raspberrypi:/home# roslaunch usb_cam usb_cam-test.launch
bash: roslaunch: command not found

```

Build seems OK, but running nodes is KO: how to fix this?

flynneva, 2023-01-15:
@fghoussen with ROS 2 roslaunch is not the command to run an executable. Instead try "ros2 run usb_cam usb_cam_exe" and it should launch.

Remember to start your docker container with "--privileged"

fghoussen, 2023-01-16:
Unfortunately still not working?!... What did I miss?

```
$ sudo docker run -it --rm --network host --privileged -e DISPLAY=:0 -v /tmp/.X11-unix:/tmp/.X11-unix ubuntu/ros2drv /bin/bash
root@raspberrypi:/home# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux 5.15.84-v8+ xhci-hcd xHCI Host Controller
Bus 001 Device 003: ID 0c45:6366 Arducam Technology Co., Ltd. Arducam OV2311 USB Camera
Bus 001 Device 002: ID 2109:3431  USB2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux 5.15.84-v8+ xhci-hcd xHCI Host Controller
root@raspberrypi:/home# 
root@raspberrypi:/home# ll /dev/video*
crw-rw---- 1 root video 81, 14 Jan 16 18:41 /dev/video0
crw-rw---- 1 root video 81, 15 Jan 16 18:41 /dev/video1
crw-rw---- 1 root video 81,  1 Jan 16 18:41 /dev/video10
crw-rw---- 1 root video 81,  2 Jan 16 18:41 /dev/video11
crw-rw---- 1 root video 81,  3 Jan 16 18:41 /dev/video12
crw-rw---- 1 root video 81,  6 Jan 16 18:41 /dev/video13
crw-rw---- 1 root video 81,  7 Jan 16 18:41 /dev/video14
crw-rw---- 1 root video 81,  8 Jan 16 18:41 /dev/video15
crw-rw---- 1 root video 81,  9 Jan 16 18:41 /dev/video16
crw-rw---- 1 root video 81,  4 Jan 16 18:41 /dev/video18
crw-rw---- 1 root video 81,  0 Jan 16 18:41 /dev/video19
crw-rw---- 1 root video 81, 10 Jan 16 18:41 /dev/video20
crw-rw---- 1 root video 81, 11 Jan 16 18:41 /dev/video21
crw-rw---- 1 root video 81, 12 Jan 16 18:41 /dev/video22
crw-rw---- 1 root video 81, 13 Jan 16 18:41 /dev/video23
crw-rw---- 1 root video 81,  5 Jan 16 18:41 /dev/video31
root@raspberrypi:/home# . /home/ros2_rolling/install/local_setup.bash
root@raspberrypi:/home# . /home/ros_workspaces/usb_cam/install/local_setup.bash
root@raspberrypi:/home# ros2 run usb_cam usb_cam_exe
No executable found
root@raspberrypi:/home# cd /home/ros_workspaces/usb_cam/
root@raspberrypi:/home/ros_workspaces/usb_cam# ros2 run usb_cam usb_cam_exe
No executable found

```

flynneva, 2023-01-17:
@fghoussen please read the README for the ros2 branch:

https://github.com/ros-drivers/usb_cam/blob/ros2/README.md

I may have misspelled the executable name.

fghoussen, 2023-01-17:
OK, I missed that!
I am trying to follow/adapt this tutorial (https://roboticsweekends.blogspot.com/2017/12/how-to-use-usb-camera-with-ros-on.html) to fetch images on master PC from USB cam plugged on raspberry.

At raspberry side (where USB cam is plugged) I get (which looks OK to me)

```
root@raspberrypi:/home# export ROS_IP="192.168.0.15" 
root@raspberrypi:/home# export ROS_MASTER_URI="http://192.168.0.1:11311" 
root@raspberrypi:/home# ros2 run usb_cam usb_cam_node_exe
[INFO] [1673988355.148383955] [usb_cam]: camera_name value: default_cam
[WARN] [1673988355.148837208] [usb_cam]: framerate: 30.000000
[INFO] [1673988355.159001868] [usb_cam]: using default calibration URL
[INFO] [1673988355.159247569] [usb_cam]: camera calibration URL: file:///root/.ros/camera_info/default_cam.yaml
[ERROR] [1673988355.159558694] [camera_calibration_parsers]: Unable to open camera calibration file [/root/.ros/camera_info/default_cam.yaml]
[WARN] [1673988355.159746729] [usb_cam]: Camera calibration file /root/.ros/camera_info/default_cam.yaml not found
[INFO] [1673988355.160023521] [usb_cam]: Starting 'default_cam' (/dev/video0) at 640x480 via mmap (yuyv, yuv422p) at 30 FPS
[INFO] [1673988356.159581619] [usb_cam]: This devices supproted formats:
[INFO] [1673988356.159876615] [usb_cam]: 	Motion-JPEG: 1600 x 1200 (50 Hz)
[INFO] [1673988356.159967465] [usb_cam]: 	Motion-JPEG: 1600 x 1200 (30 Hz)
[INFO] [1673988356.160031816] [usb_cam]: 	Motion-JPEG: 1600 x 1200 (15 Hz)
[INFO] [1673988356.160085538] [usb_cam]: 	Motion-JPEG: 1600 x 1200 (10 Hz)
[INFO] [1673988356.160137037] [usb_cam]: 	Motion-JPEG: 1280 x 960 (50 Hz)
[INFO] [1673988356.160269220] [usb_cam]: 	Motion-JPEG: 1280 x 960 (30 Hz)
[INFO] [1673988356.160324386] [usb_cam]: 	Motion-JPEG: 1280 x 960 (15 Hz)
[INFO] [1673988356.160372941] [usb_cam]: 	Motion-JPEG: 1280 x 960 (10 Hz)
[INFO] [1673988356.160416793] [usb_cam]: 	Motion-JPEG: 1280 x 960 (5 Hz)
[INFO] [1673988356.160462773] [usb_cam]: 	Motion-JPEG: 1280 x 720 (50 Hz)
[INFO] [1673988356.160507421] [usb_cam]: 	Motion-JPEG: 1280 x 720 (30 Hz)
[INFO] [1673988356.160551494] [usb_cam]: 	Motion-JPEG: 1280 x 720 (15 Hz)
[INFO] [1673988356.160636753] [usb_cam]: 	Motion-JPEG: 1280 x 720 (10 Hz)
[INFO] [1673988356.160700604] [usb_cam]: 	Motion-JPEG: 1280 x 720 (5 Hz)
[INFO] [1673988356.160758195] [usb_cam]: 	Motion-JPEG: 800 x 600 (50 Hz)
[INFO] [1673988356.160809324] [usb_cam]: 	Motion-JPEG: 800 x 600 (30 Hz)
[INFO] [1673988356.160856657] [usb_cam]: 	Motion-JPEG: 800 x 600 (15 Hz)
[INFO] [1673988356.160939230] [usb_cam]: 	Motion-JPEG: 800 x 600 (10 Hz)
[INFO] [1673988356.161035562] [usb_cam]: 	Motion-JPEG: 800 x 600 (5 Hz)
[INFO] [1673988356.161130839] [usb_cam]: 	Motion-JPEG: 640 x 480 (50 Hz)
[INFO] [1673988356.161229430] [usb_cam]: 	Motion-JPEG: 640 x 480 (30 Hz)
[INFO] [1673988356.161341225] [usb_cam]: 	Motion-JPEG: 640 x 480 (15 Hz)
[INFO] [1673988356.161457464] [usb_cam]: 	Motion-JPEG: 640 x 480 (10 Hz)
[INFO] [1673988356.161592944] [usb_cam]: 	Motion-JPEG: 640 x 480 (5 Hz)
[INFO] [1673988356.161658220] [usb_cam]: 	Motion-JPEG: 320 x 240 (50 Hz)
[INFO] [1673988356.161711664] [usb_cam]: 	Motion-JPEG: 320 x 240 (30 Hz)
[INFO] [1673988356.161759330] [usb_cam]: 	Motion-JPEG: 320 x 240 (15 Hz)
[INFO] [1673988356.161803144] [usb_cam]: 	Motion-JPEG: 320 x 240 (10 Hz)
[INFO] [1673988356.161846014] [usb_cam]: 	Motion-JPEG: 320 x 240 (5 Hz)
[INFO] [1673988356.161917828] [usb_cam]: 	Motion-JPEG: 160 x 120 (50 Hz)
[INFO] [1673988356.162030382] [usb_cam]: 	Motion-JPEG: 160 x 120 (30 Hz)
[INFO] [1673988356.162083789] [usb_cam]: 	Motion-JPEG: 160 x 120 (15 Hz)
[INFO] [1673988356.162131899] [usb_cam]: 	Motion-JPEG: 160 x 120 (10 Hz)
[INFO] [1673988356.162178936] [usb_cam]: 	Motion-JPEG: 160 x 120 (5 Hz)
[INFO] [1673988356.162224602] [usb_cam]: 	YUYV 4:2:2: 1600 x 1200 (5 Hz)
[INFO] [1673988356.162269583] [usb_cam]: 	YUYV 4:2:2: 1280 x 720 (5 Hz)
[INFO] [1673988356.162314452] [usb_cam]: 	YUYV 4:2:2: 800 x 600 (5 Hz)
[INFO] [1673988356.163028184] [usb_cam]: Timer triggering every 33 ms

```

But at master side (connected to raspberry on private LAN 192.168.0.x with static IP), I don't see any USB topic and no `image_view` node

```
root@master:/home# export ROS_IP="192.168.0.1"
root@master:/home# export ROS_MASTER_URI="http://192.168.0.1:11311"
root@master:/home# ros2 topic list
/parameter_events
/rosout
root@master:/home# ros2 run image_
image_common     image_tools      image_transport  
```

Any help / clue to finish up the tutorial (see image on master PC of a USB cam plugged on a slave raspberry) would be appreciated! Is there an updated doc on this basic use case?

flynneva, 2023-01-17:
@fghoussen you are mixing up ROS 1 and ROS 2 tutorials and variables, so be careful there 😅

With ROS 2 you shouldn't have to specify a ROS_MASTER.

ROS 2 should "just work" across multiple machines as long as they are connected to the same network and have the same ROS_DOMAIN_ID setting.

Hope this helps.

flynneva, 2023-01-17:
@fghoussen here is [a tutorial for ROS 2 across multiple machines](https://roboticsbackend.com/ros2-multiple-machines-including-raspberry-pi/) which should walk you through what you want to do.

fghoussen, 2023-01-19:
> @fghoussen you are mixing up ROS 1 and ROS 2 tutorials and variables, so be careful there sweat_smile

Right! :) No easy to find it's way across confusing docs out there!

> @fghoussen here is [a tutorial for ROS 2 across multiple machines](https://roboticsbackend.com/ros2-multiple-machines-including-raspberry-pi/) which should walk you through what you want to do.

Thanks! I'll have a look at that as soon as possible

fghoussen, 2023-01-22:
Problem seems related to the LAN config, not with the driver.

closed 2023-01-22 by fghoussen

fghoussen, 2023-01-22:
> Problem seems related to the LAN config, not with the driver.

Even more than that!... The ROS2 install doesn't allow to run hello world listener-talker example, even on my laptop alone (LAN was made of my laptop and my raspberry)?!... To install ROS2, I followed instructions from https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html in a Dockerfile (as I do not run ubuntu). Following  https://docs.ros.org/en/foxy/How-To-Guides/Installation-Troubleshooting.html does not help (on laptop, `sudo ufw allow in proto udp to\from 224.0.0.0/4` + `MULTICAST` appears on ` ifconfig docker0 `) : ` ros2 multicast receive / send  ` doesn't work?! 

Not sure where to ask for help. If people around are used to ROS2, any clue is welcome...

flynneva, 2023-01-23:
@fghoussen ROS answers is the right place to ask these sorts of questions: https://answers.ros.org/questions/

The main ROS 2 docs should also be a good resource.

If you are set on using docker, I'd recommend checking out ADE which utilizes docker volumes:

https://ade-cli.readthedocs.io/en/latest/

https://gitlab.apex.ai/apex.ai-public/ros2-ade

Good luck 🍀🫡😅

fghoussen, 2023-01-23:
Thanks for the tips!