ros-drivers/usb_cam #225: [ros2run]: Segmentation fault

wxydejoy, 2023-02-07:
I tried to modify the configuration file and run it, but I encountered the following error. I tried to modify various parameters, but it did not solve the problem. Please help me solve the problem

## Terminal

```bash
$ ros2 run usb_cam usb_cam_node_exe --ros-args --params-file ./default_cam.yaml

[INFO] [1675744515.473145126] [usb_cam]: camera_name value: test_camera
[WARN] [1675744515.473206501] [usb_cam]: framerate: 30.000000
[INFO] [1675744515.473927872] [usb_cam]: camera calibration URL: package://usb_cam/config/camera_info.yaml
[INFO] [1675744515.474210505] [usb_cam]: Starting 'test_camera' (/dev/video0) at 1280x720 via mmap (mjpeg, yuv422p) at 30 FPS
[ros2run]: Segmentation fault
```
## default_cam.yaml
```yaml
/**:
    ros__parameters:
      video_device: "/dev/video0"
      framerate: 30.0
      io_method: "mmap"
      frame_id: "camera"
      pixel_format: "mjpeg"
      color_format: "yuv422p"
      image_width: 1280
      image_height: 720
      camera_name: "test_camera"
      camera_info_url: "package://usb_cam/config/camera_info.yaml"

```

## v4l2-ctl -d /dev/video0 --list-formats-ext

```bash
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 1280x720
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)

```

wxydejoy, 2023-02-10:
I almost forgot that using yuyv is OK, but using mjpeg will report an error

wxydejoy, 2023-02-10:
Is there no maintenance here?

twdragon, 2023-02-10:
@wxydejoy I actually maintain ROS1 branch, so I am waiting now for the merge review from the ROS2 branch maintainer, @flynneva, to make many code improvements. 

For now, can you try to set `color_format: yuv420p`? The decoder may take a fault due to the matrix size mismatch

vakker, 2023-03-17:
I have the same issue. For me nothing works, except `yuyv`, even though the camera supports `mjpeg` and `h264`.
#110 might be related?

flynneva, 2023-03-18:
@wxydejoy sorry for the delay in responding here - busy few weeks for me and as typical for an open source package this is just volunteer work.

I can reproduce this error as well and have tracked it down to the `av_image_alloc` call:

[image]

Let me debug this and push a fix 👍🏼  thanks for reporting

flynneva, 2023-03-27:
@wxydejoy just an fyi. This issue lead me down the rabbit hole and I'm almost done with implementing a true fix for this. 

root cause -> the way that `usb_cam` (this repo) was handling setting of the pixel format needed to be rethought.

I should have a PR for you to try out this week. Bare with me 👍🏼

flynneva, 2023-03-27:
fyi @wxydejoy I've opened what I have so far - I havent finished implementing the MJPEG2RGB logic yet using the new pixel format class structure that #237 is adding but feel free to have a look so far. I'll finish it up this week by adding the remaining pixel formats that we used to support.

I'll ping you again once its ready for testing.

wxydejoy, 2023-03-27:
@flynneva OK ， my pleasure

flynneva, 2023-03-30:
@wxydejoy so #237 is almost ready! Right now it publishes MJPEG frames...just I'm doing something wonky with the decoding step and each channel of the decoded RGB image is not properly combined.  Feel free to take a look at [the MJPEG code](https://github.com/ros-drivers/usb_cam/blob/0619dd3f27bebbaac039e8080159ca0a644a133e/include/usb_cam/formats/mjpeg.hpp) if you'd like, it should pretty much be set...just some small tweaks are needed.

flynneva, 2023-04-01:
@wxydejoy mjpeg format should now be working on #237 if you want to try it out. All that is left on that MR is to fix linting errors and maybe add some more tests.

Note: the paramater file has changed slightly, so see that MR for the right `pixel_format` string to use:

`mjpeg2rgb` and `yuyv2rgb` are the two you want to test I think

wxydejoy, 2023-04-02:
I will test it when I have time recently

flynneva, 2023-04-02:
@wxydejoy just merged #237 to `ros2`, so just use that branch for any testing if you want to.

Once you confirm it is working as you expect I can make  a tag + release

wxydejoy, 2023-04-03:
very good 

[image]
[image]

flynneva, 2023-04-09:
@wxydejoy awesome, closing as this issue is resolved 🎉

closed 2023-04-09 by flynneva