ros-drivers/usb_cam #235: Unable to decode APP fields: Invalid data found when processing input

Hommus, 2023-03-15:
I'm receiving continuous errors in the terminal. Images are actually coming through but something is obviously not working:
```bash
[swscaler @ 0x55a6346780] No accelerated colorspace conversion found from yuv422p to rgb24.
[mjpeg @ 0x55a619f3f0] unable to decode APP fields: Invalid data found when processing input
[swscaler @ 0x55a6345f70] No accelerated colorspace conversion found from yuv422p to rgb24.
[mjpeg @ 0x55a619f3f0] unable to decode APP fields: Invalid data found when processing input
...
```
```xml
<launch>
  <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video7" />
    <param name="framerate" value="30" />
    <param name="image_width" value="1920" />
    <param name="image_height" value="1080" />
    <param name="pixel_format" value="mjpeg" />
    <param name="color_format" value="yuv422p" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="io_method" value="mmap"/>
  </node>
</launch>
```
```bash
$ v4l2-ctl --list-formats-ext -d /dev/video6
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'H264' (compressed)
	Name        : H.264
		Size: Discrete 3840x2160
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)

$ v4l2-ctl --list-formats-ext -d /dev/video7
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 640x480
			Interval: Discrete 0.067s (15.000 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
```

knorth55, 2023-03-15:
it seems your device does not support the decoding.
which CPU arch do you use?
do you use `x86` or `x86_64`?

https://github.com/FFmpeg/FFmpeg/blob/master/libswscale/yuv2rgb.c#L682-L695

Hommus, 2023-03-15:
@knorth55 I'm using a Jetson Xavier NX which I believe is aarch64.

knorth55, 2023-03-16:
im not sure but it seems ffmpeg does not support swscale.
however, we can use hardware accelerated encoder/decoder for ffmpeg in jetson, probably.

flynneva, 2023-04-02:
@Hommus doing a quick google [this is completely normal behavior from ffmpeg it looks like](https://stackoverflow.com/questions/55439184/getting-unable-to-decode-app-fields-while-playing-usb-webcam-stream-through-ff).

I don't think anything is wrong. If you'd like to stop seeing these errors just adjust the `av_set_log_level` call to `fatal` and those errors should not be printed (I've done this already on the `ros2` branch).

closed 2023-04-02 by flynneva

Hommus, 2023-04-02:
Thanks @flynneva!