ros-drivers/usb_cam #236: Error while returning frame

Hommus, 2023-03-15:
When I run a 4K stream with the following settings, I'm running into errors and dropping most of the frames:
```bash
[ INFO] [1678863198.226735952]: using default calibration URL
[ INFO] [1678863198.231955262]: camera calibration URL: file:///home/bot/.ros/camera_info/head_camera.yaml
[ INFO] [1678863198.232212160]: Unable to open camera calibration file [/home/bot/.ros/camera_info/head_camera.yaml]
[ WARN] [1678863198.232400194]: Camera calibration file /home/bot/.ros/camera_info/head_camera.yaml not found.
[ INFO] [1678863198.232630852]: Starting 'head_camera' (/dev/video6) at 3840x2160 via mmap (h264) at 15 FPS
[ WARN] [1678863198.903732819]: white_balance_temperature_auto: Input/output error
VIDIOC_S_CTRL: failed: Input/output error

[ WARN] [1678863198.919509660]: unknown control 'focus_auto'

[ERROR] [1678863201.177781502]: Error while returning frame.
[ERROR] [1678863201.353179412]: Error while returning frame.
[ERROR] [1678863201.528349352]: Error while returning frame.
[ERROR] [1678863201.693541829]: Error while returning frame.
[ERROR] [1678863201.877679336]: Error while returning frame.
...
```
```xml
<launch>
  <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video6" />
    <param name="framerate" value="15" />
    <param name="image_width" value="3840" />
    <param name="image_height" value="2160" />
    <param name="pixel_format" value="h264" />
    <param name="color_format" value="yuv420p" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="io_method" value="mmap"/>
</launch>
```
```bash
$ v4l2-ctl --list-formats-ext -d /dev/video6
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'H264' (compressed)
	Name        : H.264
		Size: Discrete 3840x2160
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)

$ v4l2-ctl --list-formats-ext -d /dev/video7
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 640x480
			Interval: Discrete 0.067s (15.000 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
```

knorth55, 2023-03-15:
can you tell me which branch or deb version you use?

can you also check if you set `yuv422p`?
if you set the color setup, is the situation like #235 ?

Hommus, 2023-03-15:
@knorth55 
- I'm building the `develop` branch.
- Using `yuv422p` results in additional errors:
```bash
[ INFO] [1678923078.265604144]: using default calibration URL
[ INFO] [1678923078.271108950]: camera calibration URL: file:///home/bot/.ros/camera_info/head_camera.yaml
[ INFO] [1678923078.271459921]: Unable to open camera calibration file [/home/bot/.ros/camera_info/head_camera.yaml]
[ WARN] [1678923078.271605966]: Camera calibration file /home/bot/.ros/camera_info/head_camera.yaml not found.
[ INFO] [1678923078.271821131]: Starting 'head_camera' (/dev/video6) at 3840x2160 via mmap (h264) at 15 FPS
[ WARN] [1678923078.960898028]: white_balance_temperature_auto: Input/output error
VIDIOC_S_CTRL: failed: Input/output error

[ WARN] [1678923078.977384065]: unknown control 'focus_auto'

[ERROR] [1678923079.464597674]: outbuf size mismatch.  pic_size: 12441600 bufsize: 16588800
[ERROR] [1678923079.775829910]: outbuf size mismatch.  pic_size: 12441600 bufsize: 16588800
[ERROR] [1678923080.020480793]: outbuf size mismatch.  pic_size: 12441600 bufsize: 16588800
[ERROR] [1678923080.141736268]: outbuf size mismatch.  pic_size: 12441600 bufsize: 16588800
[ERROR] [1678923080.270942943]: outbuf size mismatch.  pic_size: 12441600 bufsize: 16588800
[ERROR] [1678923080.499287274]: outbuf size mismatch.  pic_size: 12441600 bufsize: 16588800
[ERROR] [1678923080.694454351]: outbuf size mismatch.  pic_size: 12441600 bufsize: 16588800
[ERROR] [1678923080.877868019]: outbuf size mismatch.  pic_size: 12441600 bufsize: 16588800
[ERROR] [1678923081.057976877]: Error while returning frame.
...
```
- I'm not sure what you mean by your last question, both streams are run on the Xavier NX (aarch64) if that's what you're referring to.

knorth55, 2023-03-16:
if you got the error, it is the problem of color format.
can you try `yuv420p`?

related:
https://github.com/ros-drivers/usb_cam/issues/79

Hommus, 2023-03-16:
@knorth55 I just ran it again with `yuv420p` and I'm getting the same errors as https://github.com/ros-drivers/usb_cam/issues/236#issue-1624882398

knorth55, 2023-03-16:
im confused with #235
which arch are you using?
if you have x86(x86_64) computer, can you try the same code with x86(x86_64)?

twdragon, 2023-05-15:
It seemed `libavcodec` spent too much time decoding the H.264 frame because only CPU decoders are supported at the moment. Could you try to change only the resolution on your setup?

twdragon, 2023-05-25:
Closed due to inactivity

closed 2023-05-25 by twdragon