ros-drivers/usb_cam #243: ERROR with Rasberry pi, Video4linux: frame grabber failed

MauUnterKarottenbaum, 2023-04-12:
Dear Manager,

I am using this project with rasberry pi, this the following launch file:

_________________________________________________________________________________________
_<launch>
  <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video0" />
    <param name="image_width" value="800" />
    <param name="image_height" value="600" />
    <param name="pixel_format" value="mjpeg" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="framerate" value="300" />
    <param name="io_method" value="mmap"/>
  </node>
  <node name="image_view" pkg="image_view" type="image_view" respawn="false" output="screen">
    <remap from="image" to="/usb_cam/image_raw"/>
    <param name="autosize" value="true" />
  </node>

</launch>_
__________________________________________________________________________________________

But error occured:

__________________________________________________________________

_started roslaunch server http://raspberrypi:36869/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /usb_cam/camera_frame_id: usb_cam
 * /usb_cam/framerate: 300
 * /usb_cam/image_height: 600
 * /usb_cam/image_width: 800
 * /usb_cam/io_method: mmap
 * /usb_cam/pixel_format: mjpeg
 * /usb_cam/video_device: /dev/video0

NODES
  /
    usb_cam (usb_cam/usb_cam_node)

ROS_MASTER_URI=http://localhost:11311/

process[usb_cam-1]: started with pid [6388]
[ INFO] [1681295722.977610229]: Initializing ROS V4L USB camera 'head_camera' (/dev/video0) at 800x600 via mmap (mjpeg) at 300 FPS
[ INFO] [1681295723.007407906]: using default calibration URL
[ INFO] [1681295723.008431716]: camera calibration URL: file:///home/pi/.ros/camera_info/head_camera.yaml
[ INFO] [1681295723.009729284]: Unable to open camera calibration file [/home/pi/.ros/camera_info/head_camera.yaml]
[ WARN] [1681295723.010820538]: Camera calibration file /home/pi/.ros/camera_info/head_camera.yaml not found.
Initializing FFMPEG decoder for MJPEG compression
[ INFO] [1681295723.013967246]: Advertising std_srvs::Empty start service under name 'start_capture'
[ INFO] [1681295723.018434392]: Advertising std_srvs::Empty suspension service under name 'stop_capture'
[ INFO] [1681295723.021219379]: Advertising std_srvs::Trigger supported formats information service under name 'supported_formats'
[ INFO] [1681295723.025994616]: Advertising std_srvs::Trigger supported V4L controls information service under name 'supported_controls'
Opening streaming device /dev/video0
Video4Linux: IOCTL is not supported
Video4linux: Querying V4L2 driver for available controls (register base 0x980900, 0..99)
Sorting control names:
      exposure_auto_priority
      exposure_auto
      white_balance_temperature_auto
      brightness
      contrast
      saturation
      hue
      gamma
      gain
      power_line_frequency
      white_balance_temperature
      sharpness
      backlight_compensation
      exposure_absolute
[ WARN] [1681295723.117338304]: NOTE: the parameters generated for V4L intrinsic camera controls will be placed under namespace 'intrinsic_controls'
[ INFO] [1681295723.117538006]: Use 'intrinsic_controls/ignore' list to enumerate the controls provoking errors or the ones you just want to keep untouched
[ INFO] [1681295723.117663839]: Attempting to generate ROS parameter for V4L2 control 'exposure_auto_priority':
      Exposure, Auto Priority, min = 0, max = 1, step = 1, flags = 0x0 [0] (bool 1/0)
[ INFO] [1681295723.119802848]: Parameter intrinsic_controls/exposure_auto_priority exposed with value 0
[ INFO] [1681295723.120124809]: Attempting to generate ROS parameter for V4L2 control 'exposure_auto':
      Exposure, Auto, min = 0, max = 3, step = 1, flags = 0x0 [ 1: Manual Mode 3: Aperture Priority Mode ] [3]
[ INFO] [1681295723.122717260]: Parameter intrinsic_controls/exposure_auto exposed with value 3
[ INFO] [1681295723.123426312]: Attempting to generate ROS parameter for V4L2 control 'white_balance_temperature_auto':
      White Balance Temperature, Auto, min = 0, max = 1, step = 1, flags = 0x0 [1] (bool 1/0)
[ INFO] [1681295723.125977171]: Parameter intrinsic_controls/white_balance_temperature_auto exposed with value 1
[ INFO] [1681295723.126493687]: Attempting to generate ROS parameter for V4L2 control 'brightness':
      Brightness, min = -64, max = 64, step = 1, flags = 0x0 [-8193]
[ INFO] [1681295723.128454104]: Parameter intrinsic_controls/brightness exposed with value -8193
[ INFO] [1681295723.128784380]: Attempting to generate ROS parameter for V4L2 control 'contrast':
      Contrast, min = 0, max = 95, step = 1, flags = 0x0 [57343]
[ INFO] [1681295723.131580700]: Parameter intrinsic_controls/contrast exposed with value 57343
[ INFO] [1681295723.131835551]: Attempting to generate ROS parameter for V4L2 control 'saturation':
      Saturation, min = 0, max = 100, step = 1, flags = 0x0 [57343]
[ INFO] [1681295723.133493525]: Parameter intrinsic_controls/saturation exposed with value 57343
[ INFO] [1681295723.133740394]: Attempting to generate ROS parameter for V4L2 control 'hue':
      Hue, min = -2000, max = 2000, step = 1, flags = 0x0 [-8193]
[ INFO] [1681295723.135929773]: Parameter intrinsic_controls/hue exposed with value -8193
[ INFO] [1681295723.136637547]: Attempting to generate ROS parameter for V4L2 control 'gamma':
      Gamma, min = 100, max = 300, step = 1, flags = 0x0 [57343]
[ INFO] [1681295723.141689431]: Parameter intrinsic_controls/gamma exposed with value 57343
[ INFO] [1681295723.142093967]: Attempting to generate ROS parameter for V4L2 control 'gain':
      Gain, min = 0, max = 48, step = 1, flags = 0x0 [57343]
[ INFO] [1681295723.151731274]: Parameter intrinsic_controls/gain exposed with value 57343
[ INFO] [1681295723.152173901]: Attempting to generate ROS parameter for V4L2 control 'power_line_frequency':
      Power Line Frequency, min = 0, max = 2, step = 1, flags = 0x0 [ 0: Disabled 1: 50 Hz 2: 60 Hz ] [1]
[ INFO] [1681295723.155355757]: Parameter intrinsic_controls/power_line_frequency exposed with value 1
[ INFO] [1681295723.155683366]: Attempting to generate ROS parameter for V4L2 control 'white_balance_temperature':
      White Balance Temperature, min = 2800, max = 6500, step = 1, flags = 0x10 [57343]
[ INFO] [1681295723.158415317]: Parameter intrinsic_controls/white_balance_temperature exposed with value 57343
[ INFO] [1681295723.159184869]: Attempting to generate ROS parameter for V4L2 control 'sharpness':
      Sharpness, min = 1, max = 7, step = 1, flags = 0x0 [57343]
[ INFO] [1681295723.163717718]: Parameter intrinsic_controls/sharpness exposed with value 57343
[ INFO] [1681295723.164682417]: Attempting to generate ROS parameter for V4L2 control 'backlight_compensation':
      Backlight Compensation, min = 0, max = 1, step = 1, flags = 0x0 [57343]
[ INFO] [1681295723.167360609]: Parameter intrinsic_controls/backlight_compensation exposed with value 57343
[ INFO] [1681295723.168648473]: Attempting to generate ROS parameter for V4L2 control 'exposure_absolute':
      Exposure (Absolute), min = 1, max = 80000, step = 1, flags = 0x10 [312]
[ INFO] [1681295723.170904222]: Parameter intrinsic_controls/exposure_absolute exposed with value 312
Video4linux: Setting up auxiliary camera parameters
Video4linux: error setting camera parameter: 'Error setting controls: Invalid argument
VIDIOC_S_EXT_CTRLS: failed: Invalid argument
'
Video4linux: cannot set V4L control exposure_auto_priority
Video4linux: error setting camera parameter: 'white_balance_temperature: Invalid argument
VIDIOC_S_CTRL: failed: Invalid argument
'
Video4linux: cannot set V4L control white_balance_temperature
Video4linux: error setting camera parameter: 'exposure_absolute: Invalid argument
VIDIOC_S_EXT_CTRLS: failed: Invalid argument
'
Video4linux: cannot set V4L control exposure_absolute
[mjpeg @ 0x55a9a96ef0] error dc
[mjpeg @ 0x55a9a96ef0] error y=1 x=1
[mjpeg @ 0x55a9a96ef0] error dc
[mjpeg @ 0x55a9a96ef0] error y=17 x=49_

________________________________________________________________________________________________________

I searched a lot with google, but there is no valid solution.

could you please have a look?

Thanks a lot!

flynneva, 2023-04-13:
@MauUnterKarottenbaum if possible could you try the ros2 branch of this package? I just fixed the MJPEG implementation for it a few weeks ago now. I'm not sure what the status is for the ROS 1 side

twdragon, 2023-04-13:
@MauUnterKarottenbaum this error message means that you did not provide the `intrinsic_controls` section in the configuration, so the node tries to extract the default values of the parameters: 
- `exposure_auto_priority`
- `white_balance_temperature`
- `exposure_absolute`
and set them to the default values provided by the driver, which are not correct. To avoid this, provide the `intrinsic_controls` section in the configuration file, and the `ignore` list inside to leave the mentioned parameters intact.

Anyhow, can you share the output of the following command?
```sh
sudo v4l2-ctl --device=/dev/video0 -L
```

knorth55, 2023-04-13:
@twdragon 
I think it is better to ignore value when the default value is not found.

twdragon, 2023-04-13:
> @twdragon I think it is better to ignore value when the default value is not found.

@knorth55 the thing is they are found but are not initialized by the driver for this given device. Yes, it is better to mention them in ignore list first, and to check.

knorth55, 2023-04-13:
@twdragon i want to know the behavior, but node will stop when this happens?
if the node catch the error and keep running, i think it is good

twdragon, 2023-04-13:
> @twdragon i want to know the behavior, but node will stop when this happens? if the node catch the error and keep running, i think it is good

@knorth55 it should keep running after the errors of this kind, otherwise there is another error

knorth55, 2023-04-13:
@twdragon im releived to the behavior!

MauUnterKarottenbaum, 2023-04-13:
> @MauUnterKarottenbaum this error message means that you did not provide the `intrinsic_controls` section in the configuration, so the node tries to extract the default values of the parameters:
> 
> * `exposure_auto_priority`
> * `white_balance_temperature`
> * `exposure_absolute`
>   and set them to the default values provided by the driver, which are not correct.
> 
> To avoid this, provide the `intrinsic_controls` section in the configuration file, and the `ignore` list inside to leave the mentioned parameters intact.
> 
> Anyhow, can you share the output of the following command?
> 
> ```shell
> sudo v4l2-ctl --device=/dev/video0 -L
> ```

**Thank SK for the detailed reply. 
Issue update: Although I met the issue with rasberry pi (Debian), but it works well by using the same launch file with PC(Ubuntu 18.04) and Nvidia Jetson Nano(Ubuntu18.04). All of them are using ROS1 currently.**

**The launch file is as follows:**

`<launch>
  <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video0" />
    <param name="image_width" value="800" />
    <param name="image_height" value="600" />
    <param name="pixel_format" value="mjpeg" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="framerate" value="300" />
    <param name="io_method" value="mmap"/>
  </node>

</launch>`

**output of "sudo v4l2-ctl --device=/dev/video0 -L" are as follows:**

`                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=4 value=4
                       contrast 0x00980901 (int)    : min=0 max=95 step=1 default=36 value=36
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=32 value=32
                            hue 0x00980903 (int)    : min=-2000 max=2000 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=300 step=1 default=150 value=150
                           gain 0x00980913 (int)    : min=0 max=48 step=1 default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1
                        0: Disabled
                        1: 50 Hz
                        2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=2 value=2
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=1 value=1
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
                        1: Manual Mode
                        3: Aperture Priority Mode
              exposure_absolute 0x009a0902 (int)    : min=1 max=80000 step=1 default=312 value=312 flags=inactive
error 22 getting ext_ctrl Exposure, Auto Priority`

MauUnterKarottenbaum, 2023-04-19:
Dear, all,
Thank you very much for all the useful tips that remind me about the version related things. 
It works now.

For rasberry pi 4B user:
I explain now what i have done to make rasberry pi work with usb_cam.

1, My rasberry pi's sysstem was Debian 11 with arm64, but i installed the ros melodic(then many IOCTL, V4L2 issue appear)! Its wrong.
2, I searched the version matching stuff. There is a version mismatching related issue.  I reflashed my rasberry pi kit hardware to Debian 10 with arm64. Then configure the device with a proper sourcelist. Then i  installed ros-noetic-desktop. 
3, After that everything work.  FYI. (if you still have something related to sourcecode installing , pls use sudo apt install ros-neotic-usb-cam* . If there is ERROR msg shows like pkg not found or something wrong, pls adjust sourcelist)

closed 2023-04-24 by knorth55