ros-drivers/usb_cam #244: Running 2 different cameras using usb_cam on ROS noetic Ubuntu 20.04

stevensu1838, 2023-04-20:
Hi,

I installed usb_cam on ROS noetic Ubuntu 20.04 by **sudo apt-get install ros-noetic-usb-cam** . I've got two usb_cam connected to the PC/

$ ls /dev/video*
/dev/video0  /dev/video1  /dev/video2  /dev/video3

I can successfully run **roslaunch usb_cam usb_cam-test.launch** and get the images with no problem. 

The I created my own launch file for two webcams: 
sudo touch usb_cam_dual.launch
sudo gedit usb_cam_dual.launch

<launch>
 <group ns="camera1">
  <node name="usb_cam1" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video0" />
    <param name="image_width" value="640" />
    <param name="image_height" value="480" />
    <param name="pixel_format" value="yuyv" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="io_method" value="mmap"/>
  </node>
  <node name="image_view" pkg="image_view" type="image_view" respawn="true" output="screen">
    <remap from="image" to="/camera1/usb_cam1/image_raw"/>
    <param name="autosize" value="true" />
  </node>
 </group>

<group ns="camera2">
  <node name="usb_cam2" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video1" />
    <param name="image_width" value="640" />
    <param name="image_height" value="480" />
    <param name="pixel_format" value="yuyv" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="io_method" value="mmap"/>
  </node>
  <node name="image_view" pkg="image_view" type="image_view" respawn="true" output="screen">
    <remap from="image" to="/camera2/usb_cam2/image_raw"/>
    <param name="autosize" value="true" />
  </node>
 </group>
</launch>

knorth55, 2023-04-24:
@twdragon can you check this issue when you have time?

twdragon, 2023-04-24:
@stevensu1838 did you try to make a launchfile without `image_view`? Also, could you share the log files?

twdragon, 2023-05-15:
@stevensu1838 may it be that `/dev/video0` and `/dev/video1` are the virtual files assigned to the same camera? It sometimes happen when the kernel driver cannot recognize the exact number of frame sinks and assigns the different virtual device to the camera's hardware encoder data handler.

closed 2023-05-25 by twdragon