ros-drivers/usb_cam #246: is the library name relevant with regard to how  parameter 'video_device' is working ?

camelator, 2023-05-13:
Hi! many thanks for this great driver.
I have to admit I am a little bit challenging with this title....
but I can't find a parameter to put the usb device ID .... Instead I can set a video_device, which is not the same. 
USB ID are fixed for each component but video_device ID are not always linked to the same USB ID.

Consequently, if for a specific usb camera the /dev/videoX is changing, the library can't find the data  But my USB component is still here...
And this is exactly what is happening in my side :  I am using Ubuntu (CLI) on ARM64 machine, and each time I reboot the associated /dev/video change... and at each reboot I have to manually update the parameters file, but there are no changes in my component...

to my mind, according to the name of the library it should be a parameter for exemple usb_device with the reference of the usb component... And up to the library fo find the right /dev/video

twdragon, 2023-05-25:
It is not a ROS or this project's issue, it is a system issue. Linux assigns the device numbers to identical devices with a non-warrant policy. To avoid this, there are plenty of ways. For example, you can distinguish the devices by their serial numbers. 
https://inegm.medium.com/persistent-names-for-usb-serial-devices-in-linux-dev-ttyusbx-dev-custom-name-fd49b5db9af1
But it is a system administering task, not even a ROS-related task. Due to this, I close the issue

closed 2023-05-25 by twdragon

camelator, 2023-05-27:
I don't agree with you. This is not a system issue, because there is no place to input the USB ID but the library name is usb_cam.... Instead you are waiting for a flow id... this is not the same

twdragon, 2023-05-28:
@camelator you are not right here, sorry. This ROS drive does not care about the device bus ID or serial number from `libusb`, it connects to the V4L2 kernel driver using just a virtual filename assigned by the system, not anything other. The assignment of the virtual filename is the task performed by the kernel then. To avoid this behaviour without involving of the internal kernel API (which is also kernel version-specific) it is necessary to define the persistent device virtual filename link through `udev`, or to use ID-persistent virtual filename, for example, something like this for V4L2 device:
```
/dev/v4l/by-id/usb-CN0CVRNR7248765SAW60A02_Integrated_Webcam_HD-video-index
```

twdragon, 2023-05-28:
@camelator anyhow, it is a case to describe in the [Troubleshooting](http://wiki.ros.org/usb_cam/Troubleshooting) section