ros-drivers/usb_cam #250: [ROS2] Failed to send AVPacket to decode: Invalid data found when processing input

TAP-APIA, 2023-05-20:
```
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input

```
i got these error when i use [**ros2 launch usb_cam demo_launch.py**]

flynneva, 2023-05-21:
@TAP-APIA could you provide more info about your setup and environment? And maybe an outline of the steps you did to get this error? What pixel format are you using?

H31T0RN3RY, 2023-05-25:
i got the same error when running ros2 launch usb_cam demo_launch.py ​​command. I'm using Ubuntu 22.04 and the ROS 2 HUMBLE distro. I followed the following steps:
sudo apt-get install ros-humble-usb-cam
I got the following output:
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  ros-humble-usb-cam
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/103 kB of archives.
After this operation, 413 kB of additional disk space will be used.
Selecting previously unselected package ros-humble-usb-cam.
(Reading database ... 280557 files and directories currently installed.)
Preparing to unpack .../ros-humble-usb-cam_0.6.0-1jammy.20230426.085052_amd64.deb ...
Unpacking ros-humble-usb-cam (0.6.0-1jammy.20230426.085052) ...
Setting up ros-humble-usb-cam (0.6.0-1jammy.20230426.085052) ...
ros2 launch usb_cam demo_launch.py
I got the following output:
[INFO] [launch]: All log files can be found below /home/labiras/.ros/log/2023-05-25-09-41-03-245431-pop-os-7145
[INFO] [launch]: Default logging verbosity is set to INFO
/opt/ros/humble/share/usb_cam/config/params.yaml
[INFO] [usb_cam_node_exe-1]: process started with pid [7146]
[INFO] [show_image.py-2]: process started with pid [7148]
[usb_cam_node_exe-1] [INFO] [1685018463.514559892] [usb_cam]: camera_name value: test_camera
[usb_cam_node_exe-1] [WARN] [1685018463.514670002] [usb_cam]: framerate: 30.000000
[usb_cam_node_exe-1] [INFO] [1685018463.516225801] [usb_cam]: camera calibration URL: package://usb_cam/config/camera_info.yaml
[usb_cam_node_exe-1] [INFO] [1685018463.516768879] [usb_cam]: Starting 'test_camera' (/dev/video0) at 640x480 via mmap (mjpeg2rgb) at 30 FPS
[usb_cam_node_exe-1] [INFO] [1685018463.592343216] [usb_cam]: This devices supproted formats:
[usb_cam_node_exe-1] [INFO] [1685018463.592474649] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (30 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592490284] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (25 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592501227] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (20 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592511348] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (15 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592521578] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (10 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592531654] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (7 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592541777] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (5 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592551697] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (30 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592561485] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (25 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592571504] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (20 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592581534] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (15 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592591437] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (10 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592601049] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (7 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592610939] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (5 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592620935] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (30 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592631115] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (25 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592641127] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (20 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592650918] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (15 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592660752] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (10 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592670651] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (7 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592680421] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (5 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592690288] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (30 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592699984] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (25 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592709662] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (20 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592719388] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (15 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592729287] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (10 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592739571] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (7 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592749330] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (5 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592759112] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (30 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592776375] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (25 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592786342] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (20 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592796340] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (15 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592806297] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (10 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592816330] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (7 Hz)
[usb_cam_node_exe-1] [INFO] [1685018463.592826349] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (5 Hz)
[usb_cam_node_exe-1] unknown control 'white_balance_temperature_auto'
[usb_cam_node_exe-1] 
[usb_cam_node_exe-1] [INFO] [1685018463.595903946] [usb_cam]: Setting 'white_balance_temperature_auto' to 1
[usb_cam_node_exe-1] [INFO] [1685018463.595951181] [usb_cam]: Setting 'exposure_auto' to 3
[usb_cam_node_exe-1] unknown control 'exposure_auto'
[usb_cam_node_exe-1] 
[... cut ...]
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input

H31T0RN3RY, 2023-05-25:
how to set input format for YUV ?

H31T0RN3RY, 2023-05-25:
is there a command i can run in my terminal to change to pixel format for proper pixel ?

H31T0RN3RY, 2023-05-25:
thank you for your time and information. I'm going to change the pixel_format format to YUV like I said and see if it works. Thanks a lot in advance.

twdragon, 2023-05-25:
@flynneva as @TAP-APIA does not reply for some time, may we close this issue this weekend?

H31T0RN3RY, 2023-05-25:
/**:
    ros__parameters:
      video_device: "/dev/video0"
      framerate: 30.0
      io_method: "mmap"
      frame_id: "camera"
      pixel_format: "YUV"  # see usb_cam/supported_formats for list of supported formats
      image_width: 640
      image_height: 480
      camera_name: "test_camera"
      camera_info_url: "package://usb_cam/config/camera_info.yaml"
      brightness: -1
      contrast: -1
      saturation: -1
      sharpness: -1
      gain: -1
      auto_white_balance: true
      white_balance: 4000
      autoexposure: true
      exposure: 100
      autofocus: false
      focus: -1
i made the change in pixel_format to YUV but when running ros2 launch usb_cam demo_launch.py
I had the same error.
[INFO] [launch]: All log files can be found below /home/labiras/.ros/log/2023-05-25-10-51-34-770747-pop-os-9737
[INFO] [launch]: Default logging verbosity is set to INFO
/home/labiras/colcon_ws/install/usb_cam/share/usb_cam/config/params.yaml
[INFO] [usb_cam_node_exe-1]: process started with pid [9738]
[INFO] [show_image.py-2]: process started with pid [9740]
[usb_cam_node_exe-1] [INFO] [1685022695.028358355] [usb_cam]: camera_name value: test_camera
[usb_cam_node_exe-1] [WARN] [1685022695.028458259] [usb_cam]: framerate: 30.000000
[usb_cam_node_exe-1] [INFO] [1685022695.030517907] [usb_cam]: camera calibration URL: package://usb_cam/config/camera_info.yaml
[usb_cam_node_exe-1] [INFO] [1685022695.031097137] [usb_cam]: Starting 'test_camera' (/dev/video0) at 640x480 via mmap (mjpeg2rgb) at 30 FPS
[usb_cam_node_exe-1] [INFO] [1685022695.111225069] [usb_cam]: This devices supproted formats:
[usb_cam_node_exe-1] [INFO] [1685022695.111447566] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (30 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111468091] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (25 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111480284] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (20 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111491204] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (15 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111501854] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (10 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111512423] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (7 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111523130] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (5 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111533571] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (30 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111544268] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (25 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111554851] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (20 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111565251] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (15 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111575886] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (10 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111586456] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (7 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111597144] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (5 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111607922] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (30 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111618487] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (25 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111628927] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (20 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111647376] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (15 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111658122] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (10 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111668703] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (7 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111679540] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (5 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111690956] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (30 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111701657] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (25 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111712321] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (20 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111722996] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (15 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111733581] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (10 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111744175] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (7 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111754939] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (5 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111772997] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (30 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111783669] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (25 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111793986] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (20 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111804347] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (15 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111814769] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (10 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111825241] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (7 Hz)
[usb_cam_node_exe-1] [INFO] [1685022695.111835750] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (5 Hz)
[usb_cam_node_exe-1] unknown control 'white_balance_temperature_auto'
[... 21 lines cut ...]
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[... cut ...]
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input
[usb_cam_node_exe-1] Failed to send AVPacket to decode: Invalid data found when processing input

H31T0RN3RY, 2023-05-25:
/**:
    ros__parameters:
      video_device: "/dev/video0"
      framerate: 30.0
      io_method: "mmap"
      frame_id: "camera"
      pixel_format: "yuyv"  # see usb_cam/supported_formats for list of supported formats
      image_width: 640
      image_height: 480
      camera_name: "test_camera"
      camera_info_url: "package://usb_cam/config/camera_info.yaml"
      brightness: -1
      contrast: -1
      saturation: -1
      sharpness: -1
      gain: -1
      auto_white_balance: true
      white_balance: 4000
      autoexposure: true
      exposure: 100
      autofocus: false
      focus: -1
this appeared:
[INFO] [launch]: All log files can be found below /home/labiras/.ros/log/2023-05-25-12-17-48-717768-pop-os-16210
[INFO] [launch]: Default logging verbosity is set to INFO
/home/labiras/colcon_ws/install/usb_cam/share/usb_cam/config/params.yaml
[INFO] [usb_cam_node_exe-1]: process started with pid [16211]
[INFO] [show_image.py-2]: process started with pid [16213]
[usb_cam_node_exe-1] [INFO] [1685027868.982747141] [usb_cam]: camera_name value: test_camera
[usb_cam_node_exe-1] [WARN] [1685027868.982858213] [usb_cam]: framerate: 30.000000
[usb_cam_node_exe-1] [INFO] [1685027868.988439666] [usb_cam]: camera calibration URL: package://usb_cam/config/camera_info.yaml
[usb_cam_node_exe-1] [INFO] [1685027868.989020939] [usb_cam]: Starting 'test_camera' (/dev/video0) at 640x480 via mmap (uyvy) at 30 FPS
[usb_cam_node_exe-1] [INFO] [1685027869.066199511] [usb_cam]: This devices supproted formats:
[usb_cam_node_exe-1] [INFO] [1685027869.066452405] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (30 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066475459] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (25 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066488335] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (20 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066499557] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (15 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066510676] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (10 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066521884] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (7 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066533307] [usb_cam]: 	YUYV 4:2:2: 640 x 480 (5 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066544346] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (30 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066555577] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (25 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066566635] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (20 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066577999] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (15 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066589103] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (10 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066600353] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (7 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066611596] [usb_cam]: 	YUYV 4:2:2: 352 x 288 (5 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066623115] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (30 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066634270] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (25 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066645503] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (20 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066656981] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (15 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066668279] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (10 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066679292] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (7 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066690258] [usb_cam]: 	YUYV 4:2:2: 320 x 240 (5 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066701765] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (30 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066713114] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (25 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066724334] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (20 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066735337] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (15 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066746764] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (10 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066757996] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (7 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066769748] [usb_cam]: 	YUYV 4:2:2: 176 x 144 (5 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066780938] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (30 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066792250] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (25 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066803445] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (20 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066814427] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (15 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066825639] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (10 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066836967] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (7 Hz)
[usb_cam_node_exe-1] [INFO] [1685027869.066848606] [usb_cam]: 	YUYV 4:2:2: 160 x 120 (5 Hz)
[usb_cam_node_exe-1] unknown control 'white_balance_temperature_auto'
[usb_cam_node_exe-1] 
[... 33 lines cut ...]
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[... cut ...]
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
[show_image.py-2] unsupported encoding yuv422
Before, the usb_cam node was able to open the image, only the dark screen remained. Now after this modification the usb_cam node does not appear anymore

twdragon, 2023-05-25:
It seems your `image_view` node expects YUV420 whilst the `usb_cam` produces YUV422, so the `usb_cam` node works properly

H31T0RN3RY, 2023-05-25:
Thanks for the answer. In that case, how could I proceed so that both nodes generate the same type of format, i.e., that the usb_cam node outputs yuv 420 or that the image_view node can receive yuv 422 ?

flynneva, 2023-05-26:
@H31T0RN3RY for the ROS 2 branch of this package the pixel format string now defines both the input and desired output (they can be different if there is a conversion class implemented.

I'm on my phone now so I can't easily paste a link here, but check out the readme or the "include/usb_cam/conversions" folder on the ROS 2 branch to see which conversions are supported as of today.

There are lots of different combinations already supported so it really depends on your local HW or desired output format on which pixel format you'd want to use.

Let me know if you have any questions.

cc @twdragon

LilPetia, 2023-07-10:
I have same error, when I start use bad(cheap) 10m extension cable usb 2.0 with my usb camera. Camera with this cable work (check here https://ru.webcamtests.com/)I think it can occur because some packet lost in cable. Camera without this cable work fine with usb-cam module.
[image]
ros2 humble docker container.

flynneva, 2023-08-22:
@LilPetia could you check if #279 fixes your issue?

flynneva, 2023-08-22:
Lets continue the discussion on #274 or #275 , closing this issue

closed 2023-08-22 by flynneva