ros-drivers/usb_cam #252: Invoking "make -j2 -l2" failed

MARHEM602, 2023-05-26:
Hello, I have a problem with the package. I can gitclone the package but I have this problem when I catkin_make my work area.
I'm on Ros melodic on ubuntu 18.04 on linux. 
Thanks in advance 

[image]

twdragon, 2023-05-26:
@MARHEM602 you are compiling your workspace using by default a C++11 compiler. [`std::make_unique` is implemented in STL for C++14](https://stackoverflow.com/questions/24609271/errormake-unique-is-not-a-member-of-std). Try to add `-DCMAKE_CXX_FLAGS="-std=c++14"` in your invocation or use a newer compiler.

MARHEM602, 2023-05-26:
I updated my G++ and ran the command :
catkin_make -DCMAKE_CXX_FLAGS="-std=c++14"

But I still have the same problem

twdragon, 2023-05-26:
Incremental compiling prevents changing the standard. Try to remove `build` and `devel` subdirectories and recompile the entire workspace

MARHEM602, 2023-05-26:
[image]
I'm going to update Opencv, I'll contact you as soon as I'm done. Thank you

closed 2023-05-26 by twdragon