ros-drivers/usb_cam #253: Multiple cameras issue

HuanyuL, 2023-06-07:
Hello, I am running usb_cam node on ROS-Noetic Ubuntu20.04, I recieved this error when I have two node running at the same time. 
 ``` [ERROR] [1686147531.307026703]: Video4linux: frame grabber failed```
I don't have problem when I launch the ``usb_cam.launch``, but I seems the second node kills the first one video4linux stream. Here's my launch file.  
```<launch>
  <group ns="camera1">
    <node name="usb_cam1" pkg="usb_cam" type="usb_cam_node" output="screen">
      <rosparam command="load" file="$(find usb_cam)/config/usb_cam.yml" />
      <remap from="image" to="/camera/usb_cam1/image_raw"/>
    </node>
  </group>

  <group ns="camera2">
    <node name="usb_cam2" pkg="usb_cam" type="usb_cam_node" output="screen">
      <rosparam command="load" file="$(find usb_cam)/config/usb_cam_2.yml" />
      <remap from="image" to="/camera2/usb_cam2/image_raw"/>
    </node>
  </group>
</launch>
```
Then my first node frame mapping  timeout, the error is showing as below
```
huanyu@laptop:~/webcam_ws$ roslaunch usb_cam usb_cam_copy.launch 
... logging to /home/huanyu/.ros/log/368d6d0a-053e-11ee-8f7c-cb4ac6d7dbd1/roslaunch-laptop-16306.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://laptop:33847/

SUMMARY
========

PARAMETERS
 * /camera1/usb_cam1/camera_frame_id: head_camera
 * /camera1/usb_cam1/camera_info_url: 
 * /camera1/usb_cam1/camera_name: head_camera
 * /camera1/usb_cam1/camera_transport_suffix: image_raw
 * /camera1/usb_cam1/color_format: yuv422p
 * /camera1/usb_cam1/create_suspended: False
 * /camera1/usb_cam1/framerate: 60
 * /camera1/usb_cam1/full_ffmpeg_log: False
 * /camera1/usb_cam1/image_height: 480
 * /camera1/usb_cam1/image_width: 640
 * /camera1/usb_cam1/intrinsic_controls/exposure_absolute: 500
 * /camera1/usb_cam1/intrinsic_controls/exposure_auto: 1
 * /camera1/usb_cam1/intrinsic_controls/exposure_auto_priority: False
 * /camera1/usb_cam1/intrinsic_controls/focus_auto: False
 * /camera1/usb_cam1/intrinsic_controls/ignore: ['brightness', 'c...
 * /camera1/usb_cam1/intrinsic_controls/power_line_frequency: 1
 * /camera1/usb_cam1/intrinsic_controls/white_balance_temperature_auto: False
 * /camera1/usb_cam1/io_method: mmap
 * /camera1/usb_cam1/pixel_format: yuyv
 * /camera1/usb_cam1/start_service_name: start_capture
 * /camera1/usb_cam1/stop_service_name: stop_capture
 * /camera1/usb_cam1/video_device: /dev/video2
 * /camera2/usb_cam2/camera_frame_id: head_camera
 * /camera2/usb_cam2/camera_info_url: 
 * /camera2/usb_cam2/camera_name: head_camera
 * /camera2/usb_cam2/camera_transport_suffix: image_raw
 * /camera2/usb_cam2/color_format: yuv422p
 * /camera2/usb_cam2/create_suspended: False
 * /camera2/usb_cam2/framerate: 60
 * /camera2/usb_cam2/full_ffmpeg_log: False
 * /camera2/usb_cam2/image_height: 480
 * /camera2/usb_cam2/image_width: 640
 * /camera2/usb_cam2/intrinsic_controls/exposure_absolute: 100
 * /camera2/usb_cam2/intrinsic_controls/exposure_auto: 1
 * /camera2/usb_cam2/intrinsic_controls/exposure_auto_priority: False
 * /camera2/usb_cam2/intrinsic_controls/focus_auto: False
 * /camera2/usb_cam2/intrinsic_controls/ignore: ['brightness', 'c...
[... 185 lines cut ...]
[ INFO] [1686147526.258369665]: Parameter intrinsic_controls/focus_absolute exposed with value 220
[ INFO] [1686147526.258382027]: Attempting to generate ROS parameter for V4L2 control 'zoom_absolute':
	Zoom, Absolute, min = 0, max = 3, step = 1, flags = 0x0 [0]
[ INFO] [1686147526.258513748]: Parameter intrinsic_controls/pan_absolute exposed with value 0
[ INFO] [1686147526.258528345]: Attempting to generate ROS parameter for V4L2 control 'tilt_absolute':
	Tilt (Absolute), min = -43200, max = 43200, step = 3600, flags = 0x0 [0]
[ INFO] [1686147526.258655177]: Parameter intrinsic_controls/zoom_absolute exposed with value 0
[ INFO] [1686147526.258800726]: Parameter intrinsic_controls/tilt_absolute exposed with value 0
[ INFO] [1686147526.258814834]: Attempting to generate ROS parameter for V4L2 control 'focus_absolute':
	Focus (absolute), min = 120, max = 960, step = 4, flags = 0x0 [220]
[ INFO] [1686147526.259069197]: Parameter intrinsic_controls/focus_absolute exposed with value 220
[ INFO] [1686147526.259082537]: Attempting to generate ROS parameter for V4L2 control 'zoom_absolute':
	Zoom, Absolute, min = 0, max = 3, step = 1, flags = 0x0 [0]
Video4linux: Setting up auxiliary camera parameters
[ INFO] [1686147526.259345071]: Parameter intrinsic_controls/zoom_absolute exposed with value 0
Video4linux: Setting up auxiliary camera parameters
Video4linux: frame mapping timeout (11)
[ERROR] [1686147531.307026703]: Video4linux: frame grabber failed
Video4linux: frame mapping timeout (11)
[ERROR] [1686147536.312275487]: Video4linux: frame grabber failed
```
I tried to rosnode kill the ```/camera2/usb_cam2``` node then the first node coming back immediately. I am using two [trust 23637 camera](https://www.trust.com/es/product/23637-tyro-full-hd-webcam), all the configs in yaml file are supported my this camera. The strange thing  is that this issue can be solved by replace one camera by a different model, here I replaced it by logitech c920.

Could you please have a look?
Many thanks!

twdragon, 2023-06-07:
@HuanyuL did you try to run the nodes accessing the cameras by ID but not via the `/dev/video`? Could you please test this access mode to ensure it is not a kernel driver problem?

HuanyuL, 2023-06-13:
> 

hello, It's not a kernel driver issue, I sovled by reduce the fps to 10, my usb bandwidth dosen't support two camera streaming at the same time. Thank you for suggestion!

flynneva, 2023-10-28:
Closing as resolved

closed 2023-10-28 by flynneva