ros-drivers/usb_cam #255: [Community assistance needed] Merging ROS 1 and ROS 2 code over the same internal API

twdragon, 2023-06-08:
This issue is created to encourage any assistance in merging ROS1 and ROS2 branches. I want also to kindly ask @flynneva as a maintainer of ROS2 branch to write here the links to his studies about the merge, especially about CMake solution he uses. Please feel free to share your solutions. The huge work is already done, so the merging process should start!

#### Current general milestones
- Merging CMake build scenarios (waiting for a solution from @flynneva)
- Dynamic generation of ROS parameters during node startup (solved for ROS1, searching for a solution for ROS2)
- Exposure of ROS node as just a wrapper over the internal driver library API

flynneva, 2023-06-11:
@twdragon so the basic idea for this was already outlined in #203.

We can take the work I did last year, which is [still available here](https://github.com/flynneva/usb_cam/pull/2) and update the CI to work with the code that's been updated since then.

I'll put together a quick PR tomorrow and see what else needs to be updated.

flynneva, 2023-11-05:
@twdragon @knorth55 fyi I've merged the CI for both ROS 1 and ROS 2 into the `ros2` branch and set the `ros1` jobs to be allowed to fail with #294.

Now it is up to you or someone else in the community to finish updating the `ros1` node on the `ros2` branch. When you do we can remove that "allow failure" flag for the CI job 👍🏼

flynneva, 2025-02-09:
Closing as wont complete, since ROS 1 is being deprecated in a few months anyways

closed 2025-02-09 by flynneva