ros-drivers/usb_cam #259: images published out of order

MFernandezCarmona, 2023-07-06:
Hi,
I'm using `usb_cam_node_exe` node in Humble and `image_raw` messages are being published out of (temporal) order. 

Steps to reproduce:

1. Plug usb camera (tried two different ones)

2. Run  `usb_cam_node_exe` node

```console
ros2 run usb_cam usb_cam_node_exe --ros-args -r __ns:=/camera
```

3.Create a simple python node to check image timestamps:
```python
import rclpy
from rclpy.node import Node
from sensor_msgs.msg import Image
from rclpy.time import Time

class ImageTimestampSubscriber(Node):
    def __init__(self):
        super().__init__('image_timestamp_subscriber')
        self.subscription = self.create_subscription(
            Image,
            '/camera/image_raw',
            self.image_callback,
            10
        )       

    def image_callback(self, msg):
        if hasattr(self,'last_time'):
            time_diff = Time.from_msg(msg.header.stamp) - self.last_time
            if time_diff.nanoseconds<0.0:
                self.get_logger().info(f" image ts dif: {time_diff.nanoseconds/1e6} ms")

        self.last_time = Time.from_msg(msg.header.stamp)

def main(args=None):
    rclpy.init(args=args)
    node = ImageTimestampSubscriber()
    rclpy.spin(node)
    node.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```
4.Python script output shows that some messages are not in order:
```console
[INFO] [1688644345.950392161] [image_timestamp_subscriber]:  image ts dif: -964.004 ms
[INFO] [1688644346.950561701] [image_timestamp_subscriber]:  image ts dif: -967.999 ms
[INFO] [1688644347.958506978] [image_timestamp_subscriber]:  image ts dif: -967.996 ms
[INFO] [1688644348.930496476] [image_timestamp_subscriber]:  image ts dif: -964.003 ms
[INFO] [1688644349.938465496] [image_timestamp_subscriber]:  image ts dif: -967.999 ms
[INFO] [1688644350.942504378] [image_timestamp_subscriber]:  image ts dif: -968.0 ms
[INFO] [1688644351.950476913] [image_timestamp_subscriber]:  image ts dif: -967.996 ms
[INFO] [1688644352.958440612] [image_timestamp_subscriber]:  image ts dif: -964.004 ms
```
Any ideas on what might be wrong?
Thanks

javieryu, 2023-07-10:
I am also getting negative delays on my setup.
- Ubuntu 22.04.2
- Raspberry Pi 4B
- ROS Iron

I replicate with:
`ros2 run usb_cam usb_cam_node_exe --ros-args --parms-file params.yaml`.

Params file:
```
/**:
    ros__parameters:
      video_device: "/dev/video0"
      framerate: 20.0
      io_method: "mmap"
      frame_id: "camera"
      pixel_format: "mjpeg2rgb"  # see usb_cam/supported_formats for list of supported formats
      image_width: 1280
      image_height: 720
      camera_name: "test_camera"
      camera_info_url: "package://usb_cam/config/camera_info.yaml"
      brightness: 42
      contrast: 29
      saturation: 91
      sharpness: 3
      gain: 0
      auto_white_balance: false
      white_balance: 4000
      autoexposure: false
      exposure: 100
      autofocus: false
      focus: -1
```

Results from `ros2 topic delay /image_raw` then show a negative delay between messages. 
[image]

javieryu, 2023-07-10:
Just created a pull request that fixes this issue. @MFernandezCarmona You just need to remove the round operation from the places where time stamps are computed.

MFernandezCarmona, 2023-07-10:
Thanks! I'll give it a try

boitumeloruf, 2023-07-21:
The sole removal of the rounding in calculating the timestamp does not resolve the problem. I believe, the problem arose due to missing handling of sub-second accuracy in the calculation of the epoch time shift. See my PR #267 for more details.

flynneva, 2023-07-23:
Does anyone want to volunteer to do some testing to see if this issue is now fixed?

MFernandezCarmona, 2023-07-27:
Hi,
Just tested PR #267 and it shows a consistent and positive delay of around 74ms. I'd say it's solved, unless there is any other complex testing you would like to try.
Thanks!

closed 2023-07-27 by MFernandezCarmona