ros-drivers/usb_cam #260: Auto exposure and other parameter names

javieryu, 2023-07-06:
Some of the parameters for my device seem to have names that are slightly different than those set using this library. For instance, the ros parameter for `autoexposure` is set using the device control `exposure_auto`, but for my device it is `auto_exposure` according to `v4l2-ctl --device=0 --all`. This seems to result in a outcome where the `autoexposure` setting does not change anything when I set it in the params file or with `ros2 param set ...`.

Output from `v4l2-ctl --device=0 --all`:
[image]

Result when running `ros2 run usb_cam usb_cam_node_exe --ros-args --params-file params.yaml`:
[image]

It seems like the naming conventions that is different for my camera/drivers? So far my workaround is to use the `v4l2-ctl --device=0 --set-ctrl=...` command  after running the usb_cam node. Is there a better way to do this?

flynneva, 2023-07-13:
@javieryu thanks for reporting this! I'll look into it and circle back this weekend.

javieryu, 2023-07-13:
Great, I forked and changed the parameter name strings in `UsbCamNode::set_v4l2_params()` to match the outputs of `v4l2-ctl`, and that seems to have fixed my problems. I'm assuming that these names are different on each camera depending on the drivers. 

For reference here are the camera specs:
[image]

Is the plan to eventually automatically discover the camera parameter names?

TZECHIN6, 2023-08-16:
Seem i am also running into this as well:

```
User Controls

                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=50
                       contrast 0x00980901 (int)    : min=0 max=95 step=1 default=0 value=0
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=64 value=64
                            hue 0x00980903 (int)    : min=-2000 max=2000 step=1 default=0 value=0
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=300 step=1 default=100 value=100
                           gain 0x00980913 (int)    : min=1 max=8 step=1 default=1 value=1
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=1 (50 Hz)
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=2 value=2
         backlight_compensation 0x0098091c (int)    : min=0 max=3 step=1 default=3 value=3

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3 (Aperture Priority Mode)
         exposure_time_absolute 0x009a0902 (int)    : min=9 max=625 step=1 default=157 value=157 flags=inactive
     exposure_dynamic_framerate 0x009a0903 (bool)   : default=0 value=1
```

my `v4l2-ctl` version is `1.22.1`

flynneva, 2023-08-16:
> Is the plan to eventually automatically discover the camera parameter names?

@javieryu this would be an awesome awesome feature! Just not sure how to implement it 😅 PR's / implementation ideas welcome 🙏🏼

twdragon, 2023-08-16:
@flynneva @javieryu I am working now on it, as it was already implemented in ROS 1 branch. Hope to complete and create a PR soon!

TZECHIN6, 2023-08-16:
@twdragon thanks for your contribution, I look forward to your PR

nickswalker, 2023-08-29:
Just hit this too. It would be nice for parameters that fail to set to print to `ROS_ERROR`. I missed the error output from the v4l2-ctl command because I was running the node as part of a launch file. 

We hardcoded our parameters in a fork here for the time being: https://github.com/hcrlab/usb_cam/

Our camera's parameters (teleop fisheye cameras from Hello Robot Stretch, in case others are searching):
````
User Controls

                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=50
                       contrast 0x00980901 (int)    : min=0 max=100 step=1 default=40 value=100
                     saturation 0x00980902 (int)    : min=0 max=128 step=1 default=64 value=128
                            hue 0x00980903 (int)    : min=-180 max=180 step=1 default=0 value=0
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=500 step=1 default=300 value=100
                           gain 0x00980913 (int)    : min=0 max=128 step=1 default=64 value=10
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=2 (60 Hz)
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=10 default=4600 value=4200 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=100 step=1 default=56 value=45
         backlight_compensation 0x0098091c (int)    : min=0 max=2 step=1 default=0 value=0

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3 (Aperture Priority Mode)
         exposure_time_absolute 0x009a0902 (int)    : min=1 max=10000 step=1 default=166 value=10000 flags=inactive
     exposure_dynamic_framerate 0x009a0903 (bool)   : default=0 value=0
                   pan_absolute 0x009a0908 (int)    : min=-57600 max=57600 step=3600 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=-43200 max=43200 step=3600 default=0 value=0
                  zoom_absolute 0x009a090d (int)    : min=0 max=3 step=1 default=0 value=0
````

AlexanderSalge, 2025-04-16:
any updates on this issue?