ros-drivers/usb_cam #262: Memory leak - ros2 humble

firesurfer, 2023-07-12:
# Issue

usb_cam node allocates memory over time without freeing it until the systems hangs and the oom kills it.

```
[20524.624725] Out of memory: Killed process 267615 (usb_cam_node_ex) total-vm:31232704kB, anon-rss:27694492kB, file-rss:12320kB, shmem-rss:0kB, UID:1000 pgtables:59852kB oom_score_adj:0

```

# System

Ubuntu 22.04
ROS2 Humble (binary install)
usb_cam: 0.6.0

v4l2-ctl -D
```
Driver Info:
	Driver name      : uvcvideo
	Card type        : HD Pro Webcam C920
	Bus info         : usb-0000:0c:00.3-2
	Driver version   : 5.19.17
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : HD Pro Webcam C920
	Serial           : 
	Bus info         : usb-0000:0c:00.3-2
	Media version    : 5.19.17
	Hardware revision: 0x0000000c (12)
	Driver version   : 5.19.17
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : HD Pro Webcam C920
	Function         : V4L2 I/O
	Flags            : default
	Pad 0x01000007   : 0: Sink
	  Link 0x0200001c: from remote pad 0x100000a of entity 'Processing 3' (Video Pixel Formatter): Data, Enabled, Immutable
```

##  Camera yaml:

```camera_1/camera_1_node:
    ros__parameters:
      video_device: "/dev/video0"
      framerate: 30.0
      io_method: "mmap"
      frame_id: "camera_1"
      pixel_format: "mjpeg2rgb"
#      color_format: "yuv422p" #Will result in less then 30fps on uncompressed topic
      image_width: 1920
      image_height: 1080
      camera_name: "logitech_c920"
      camera_info_url: "package://winder_camera/config/camera_1_info.yaml"
```

## Further info

Camera image is displayed in rviz

Let me know if you need futher information

flynneva, 2023-07-13:
@firesurfer thanks for reporting. I'll dig into this over the weekend when I have some time 👍🏼

flynneva, 2023-07-15:
@firesurfer I've addressed the major memory leak issues with #264 (or at least a lot of them 😅 )  There are still a few smaller ones left, but I've run out of time today and need to work on some other things this weekend.

Hopefully those fixes are "good enough" for you for now - I'll leave this issue open even after #264 is merged to address the rest of the memory issues later on.

flynneva, 2023-07-23:
@firesurfer I believe I've addressed the last memory leak related to this drivers code in #268. There are still 3 memory leaks reported by asan but those are from lower-level ROS 2 packages I believe and not something I can address here in this repo.

I've added a `SANITIZE` option to the CMakelists.txt file as well so others can quickly and easily address memory leaks too.

Maybe down the road I can add a CI job that uses this option to auto-detect memory leaks in PRs

firesurfer, 2023-07-24:
@flynneva I just tested #268 
It looks in general way better but still starts consuming memory without freeing it (at least way slower than before).
The interesting thing is: I had to subscribe to the raw topic (didnt try the other ones) for this to happen. 

I ran valgrind with leak-check=yes

```
==14672== Invalid free() / delete / delete[] / realloc()
==14672==    at 0x484B27F: free (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==14672==    by 0x7B99565: av_buffer_unref (in /usr/lib/x86_64-linux-gnu/libavutil.so.56.70.100)
==14672==    by 0x693C098: av_packet_unref (in /usr/lib/x86_64-linux-gnu/libavcodec.so.58.134.100)
==14672==    by 0x6703ABA: std::_Sp_counted_ptr_inplace<usb_cam::formats::MJPEG2RGB, std::allocator<usb_cam::formats::MJPEG2RGB>, (__gnu_cxx::_Lock_policy)2>::_M_dispose() (in /home/user/workspace/webcam_ws/install/usb_cam/lib/libusb_cam.so)
==14672==    by 0x6701731: usb_cam::UsbCam::~UsbCam() (in /home/user/workspace/webcam_ws/install/usb_cam/lib/libusb_cam.so)
==14672==    by 0x6605DA3: usb_cam::UsbCamNode::~UsbCamNode() (in /home/user/workspace/webcam_ws/install/usb_cam/lib/libusb_cam_node.so)
==14672==    by 0x10C4DE: main (in /home/user/workspace/webcam_ws/install/usb_cam/lib/usb_cam/usb_cam_node_exe)
==14672==  Address 0x1b9c5000 is in a rw- mapped file /dev/video0 segment
==14672== 
==14672== 
==14672== HEAP SUMMARY:
==14672==     in use at exit: 161,055,721 bytes in 4,224 blocks
==14672==   total heap usage: 163,301 allocs, 159,078 frees, 1,892,940,843 bytes allocated
==14672== 
==14672== 16 bytes in 1 blocks are possibly lost in loss record 122 of 476
==14672==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==14672==    by 0x4013E4D: malloc (rtld-malloc.h:56)
==14672==    by 0x4013E4D: allocate_dtv_entry (dl-tls.c:684)
==14672==    by 0x4013E4D: allocate_and_init (dl-tls.c:709)
==14672==    by 0x4013E4D: tls_get_addr_tail (dl-tls.c:907)
==14672==    by 0x401826B: __tls_get_addr (tls_get_addr.S:55)
==14672==    by 0x1980A147: ???
==14672==    by 0x40021A1: call_destructors (dl-close.c:129)
==14672==    by 0x4E68C84: _dl_catch_exception (dl-error-skeleton.c:182)
==14672==    by 0x4002635: _dl_close_worker.part.0.isra.0 (dl-close.c:292)
==14672==    by 0x40032A1: _dl_close_worker (dl-close.c:150)
==14672==    by 0x40032A1: _dl_close (dl-close.c:818)
==14672==    by 0x4E68C27: _dl_catch_exception (dl-error-skeleton.c:208)
==14672==    by 0x4E68CF2: _dl_catch_error (dl-error-skeleton.c:227)
==14672==    by 0x4D841AD: _dlerror_run (dlerror.c:138)
==14672==    by 0x4D83ED7: dlclose@@GLIBC_2.34 (dlclose.c:31)
==14672== 
==14672== 64 bytes in 1 blocks are definitely lost in loss record 225 of 476
==14672==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==14672==    by 0x6703F09: usb_cam::formats::MJPEG2RGB::MJPEG2RGB(int const&, int const&) (in /home/user/workspace/webcam_ws/install/usb_cam/lib/libusb_cam.so)
==14672==    by 0x670489C: usb_cam::UsbCam::set_pixel_format_from_string(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/user/workspace/webcam_ws/install/usb_cam/lib/libusb_cam.so)
==14672==    by 0x67014B8: usb_cam::UsbCam::configure(usb_cam::parameters_t&, usb_cam::utils::io_method_t const&) (in /home/user/workspace/webcam_ws/install/usb_cam/lib/libusb_cam.so)
==14672==    by 0x660A04D: usb_cam::UsbCamNode::init() (in /home/user/workspace/webcam_ws/install/usb_cam/lib/libusb_cam_node.so)
==14672==    by 0x660BE6A: usb_cam::UsbCamNode::UsbCamNode(rclcpp::NodeOptions const&) (in /home/user/workspace/webcam_ws/install/usb_cam/lib/libusb_cam_node.so)
==14672==    by 0x661399D: rclcpp_components::NodeFactoryTemplate<usb_cam::UsbCamNode>::create_node_instance(rclcpp::NodeOptions const&) (in /home/user/workspace/webcam_ws/install/usb_cam/lib/libusb_cam_node.so)
==14672==    by 0x10BDB4: main (in /home/user/workspace/webcam_ws/install/usb_cam/lib/usb_cam/usb_cam_node_exe)
==14672== 
==14672== 128 bytes in 2 blocks are definitely lost in loss record 360 of 476
==14672==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==14672==    by 0x400F21D: malloc (rtld-malloc.h:56)
==14672==    by 0x400F21D: resize_scopes (dl-open.c:295)
==14672==    by 0x400F21D: dl_open_worker_begin (dl-open.c:707)
==14672==    by 0x4E68C27: _dl_catch_exception (dl-error-skeleton.c:208)
==14672==    by 0x400DF99: dl_open_worker (dl-open.c:782)
==14672==    by 0x4E68C27: _dl_catch_exception (dl-error-skeleton.c:208)
==14672==    by 0x400E34D: _dl_open (dl-open.c:883)
==14672==    by 0x4D846BB: dlopen_doit (dlopen.c:56)
==14672==    by 0x4E68C27: _dl_catch_exception (dl-error-skeleton.c:208)
==14672==    by 0x4E68CF2: _dl_catch_error (dl-error-skeleton.c:227)
==14672==    by 0x4D841AD: _dlerror_run (dlerror.c:138)
==14672==    by 0x4D84747: dlopen_implementation (dlopen.c:71)
==14672==    by 0x4D84747: dlopen@@GLIBC_2.34 (dlopen.c:81)
==14672==    by 0x4A9D8CB: rcutils_load_shared_library (in /opt/ros/humble/lib/librcutils.so)
==14672== 
==14672== 143 (64 direct, 79 indirect) bytes in 1 blocks are definitely lost in loss record 366 of 476
==14672==    at 0x4849013: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==14672==    by 0x50E3A23: ??? (in /opt/ros/humble/lib/librosidl_typesupport_cpp.so)
[... 267 lines cut ...]
[... cut ...]
==14672==    indirectly lost: 156,920,038 bytes in 1,868 blocks
==14672==      possibly lost: 3,800,475 bytes in 19 blocks
==14672==    still reachable: 292,048 bytes in 1,084 blocks
==14672==         suppressed: 0 bytes in 0 blocks
==14672== Reachable blocks (those to which a pointer was found) are not shown.
==14672== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==14672== 
==14672== For lists of detected and suppressed errors, rerun with: -s
==14672== ERROR SUMMARY: 23 errors from 23 contexts (suppressed: 0 from 0)

```

flynneva, 2023-07-25:
@firesurfer ah you are right! I only tested without subscribing. Let me try again with subscribing and see what else I can address 👍

flynneva, 2023-07-25:
@firesurfer well I cant seem to get valgrind to work for me to show the same exact memory leaks as you, but I think from your output I'm able to fix the leaks (fingers crossed). I just pushed one more commit to #268 so give it a go and let me know what you get.

Could you share with me how you run valgrind? I've tried the following but I dont see the same memory leaks as you:

```
 valgrind --leak-check=full \
   --show-leak-kinds=all \
   --track-origins=yes  \
   --verbose \
   --log-file=valgrind-out.txt ./install/usb_cam/lib/usb_cam/usb_cam_node_exe 
```

flynneva, 2023-07-25:
@firesurfer oh nevermind 😓 I was using `yuyv` instead of `mjpeg`. Ignore my stupidity 😅

flynneva, 2023-07-25:
@firesurfer I was able to track down _most_ of the issues in the `mjpeg2rgb` conversion I think. Now that I can reproduce locally though I should be able to address the rest of them over the weekend

flynneva, 2023-07-25:
@firesurfer actually wasn't able to solve this just yet. Give me a few days and ill track it dow!

firesurfer, 2023-08-14:
@flynneva hope you are doing fine! Did you manage to dig further into the issue by now ?

flynneva, 2023-08-24:
@firesurfer still working on it a bit and plan to do more work over the weekend.

I'll keep you posted 👍

flynneva, 2023-08-30:
@firesurfer small update here: I've narrowed the major leakages down to the `av_packet_from_data` call. We'll need to come up with an alternative way of initializing the ffmpeg AVPacket, or figuring out what is leaking there.

Sorry this has taken so long...I had a friends wedding last week and things have been pretty busy for me the last few weeks 😅

firesurfer, 2023-08-30:
@flynneva  No worries. Thanks for maintaining this package!

I just took a quick look myself into the code and the FFmpeg documentation. 

So if I understand the docs correctly (https://ffmpeg.org/doxygen/2.7/group__lavc__packet.html#ga7ca877e1f0ded89a27199b65e9a077dc) One should only pass a data pointer which has been allocated with av_malloc. Which isn't the case here if I am right. 

See also this stackoverflow question:
https://stackoverflow.com/questions/63617241/how-to-load-data-to-an-ffmpeg-avbuffer

flynneva, 2023-08-31:
@firesurfer I think I got it: https://github.com/ros-drivers/usb_cam/pull/279/commits/bd6bb91c7279819ad4136001b5e5a138dffa5452

now the only leaks that are left are outside of this package it looks like

firesurfer, 2023-09-04:
@flynneva Sounds great. I hope I find some time this week to test it!

flynneva, 2023-09-08:
@firesurfer just an fyi the update is now on the main `ros2` branch, so just use that one for your tests 👍🏼

firesurfer, 2023-09-18:
@flynneva  Just tested the latest ros-iron release and it seems to fix the issue!
Thanks a lot

flynneva, 2023-09-18:
@firesurfer awesome thanks for confirming! I'll go ahead and close this issue then 👍

closed 2023-09-18 by flynneva