ros-drivers/usb_cam #266: Framerate issue - 30fps not reached at full hd - ros humble

firesurfer, 2023-07-17:
# Issue

On the `image_raw` and the compressed topic the desired framerate of 30 fps is not even closely reached.  On average I reach with the default configuration around 16 fps.

Interestingly the update rate on the raw/compressed/info topics does not always match. But this could be a measuring issue. 

The Webcam reports (also in the startup of the usb cam node) that it supports Full HD at 30Hz. Using VLC media player I can also play it at 30fps (if the media information displayed is correct)

# System

Ubuntu 22.04
ROS2 Humble (binary install)
usb_cam: 0.6.0

v4l2-ctl -D

```
Driver Info:
	Driver name      : uvcvideo
	Card type        : HD Pro Webcam C920
	Bus info         : usb-0000:0c:00.3-2
	Driver version   : 5.19.17
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : HD Pro Webcam C920
	Serial           : 
	Bus info         : usb-0000:0c:00.3-2
	Media version    : 5.19.17
	Hardware revision: 0x0000000c (12)
	Driver version   : 5.19.17
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : HD Pro Webcam C920
	Function         : V4L2 I/O
	Flags            : default
	Pad 0x01000007   : 0: Sink
	  Link 0x0200001c: from remote pad 0x100000a of entity 'Processing 3' (Video Pixel Formatter): Data, Enabled, Immutable
```

# Camera yaml

    ros__parameters:
      video_device: "/dev/video0"
      framerate: 30.0
      io_method: "mmap"
      frame_id: "camera_1"
      pixel_format: "mjpeg2rgb"
#      color_format: "yuv422p" #Will result in less then 30fps on uncompressed topic
      image_width: 1920
      image_height: 1080
      camera_name: "logitech_c920"
      camera_info_url: "package://camera/config/camera_1_info.yaml"

# Further info

I played around with the following parameters:

1. Based on a comment in the main loop of the usb cam node I set a very small manual exposure time. This did not increase the framerate/update rate
2. Selecting a `userptr` as io_method. Did increase the framerate by like 2 fps
3. Reducing the image size from Full HD solved the issue. The frames are then published at 30fps
4. color_format. See my comment in the camera yaml file. Setting an explicit color_format has a negative impact on the frame rate. The difference compressed/raw topic publish rate seemed to be higher.

flynneva, 2023-07-21:
@firesurfer thanks for reporting this! So I've been able to track it down to [the `select()` call](https://github.com/ros-drivers/usb_cam/blob/ros2/src/usb_cam.cpp#L668). For some reason this is taking a very long time it looks like. I'll keep digging into it and provide a fix, apologies for the issue!

flynneva, 2023-07-22:
@firesurfer did some more testing today after the timestamp update with #267 now merged to `ros2` and it looks like only the `image_raw` topic reports a lower-than-normal hz with ros2 topic hz now. The `image_raw/theora` and `image_compressed` both output the expected / proper hz for me.

Also that `select` issue I was seeing is no longer there either, and now it seems to take the "expected" period of ~33ms for 30Hz and adjusts based on the selected framerate.

Can you confirm this is also what you are seeing?

firesurfer, 2023-07-24:
@flynneva #267 definitely seemed to bring an improvement. With a reduced image size also the image_raw topic reports 30Hz

evan-flynn-apexai, 2023-07-24:
@firesurfer I'll go ahead and close this issue then as it seems to be resolved.

flynneva, 2023-07-24:
@firesurfer I'll go ahead and close this issue then 👍🏼

firesurfer, 2023-10-04:
@flynneva Sorry to bring up this issue again. I just tested again with the latest ros2-iron release and I run into the same issue. Camera works fine with vlc at 1080p@30Fps, with the usb_cam_node I can only reach around 15-20fps. 
Reducing to 720p solves the issue. Configuration is still the same as above.

flynneva, 2023-10-06:
@firesurfer how are you measuring your FPS? Via `ros2 topic hz`? Which topic are you testing with? Compressed or uncompressed?

I am not able to reproduce your issue with the compressed topic at full resolution for my camera (1280 x 720) at 30 fps:

[image]

firesurfer, 2023-10-06:
I checked the FPS in two different ways:
1. ros2 topic hz -> uncompressed tpic
2. I have a node which writes video streams to files -> The resulting framerate in the file was also below 30fps

The issue only occurs at

```
  image_width: 1920
  image_height: 1080
```

If I reduce it to

```
      image_width: 1280
      image_height: 720
```

the framerate is correct (both types of measurement)

firesurfer, 2023-10-09:
@flynneva I just ran some tests.

1. 1280 x 720 : 25% cpu load per node, 30fps (ros2 topic hz) -> ros2 topic hz subscriptions increased cpu load by 10%
2. 1920x 1080: 40% cpu load per node, 16fps
In both cases i ran two camera nodes and subscribed to the image topics via rviz

As reference with vlc at 1920x1080
30-40% cpu at 30fps

flynneva, 2023-10-10:
@firesurfer awesome stuff! Ok so definitely doesn't look like you are topping out anything...give me a few days to debug a bit and circle back. Thanks again for reporting this and providing more info 🙏🏼

flynneva, 2023-10-17:
@firesurfer generated a quick flamegraph to see where we can save the most time:

[image]

Probably a little hard to see, but looks like most of the time is spent in `libavcodec` calls. I'll look into where we might be going wrong with the avcodec calls and see where we can improve 👍🏼

flynneva, 2023-10-17:
@firesurfer looks like we might need to change a few things:

1. In the avcodec space:
    1. Might need to adjust the `avcodec_context` settings to be [multithreaded](https://stackoverflow.com/questions/55186822/whats-ffmpeg-doing-with-avcodec-send-packet)
    2. That link also says that maybe calling `av_send_packet` multiple times would help...but I'm not sure
 1. ROS 2 world: we could add a custom `main` and use the MultithreadedExecutor

firesurfer, 2023-10-21:
@flynneva I guess working on 1. makes the most sense. I also ran a callgrind analysis and also came to the conclusion that `av_send_packet` is where most of the time is spent

flynneva, 2023-10-30:
@flynneva note to self: this PR / issue looks promising:

https://github.com/Kitware/kwiver/pull/1504

Do some testing this week to see if sending a nullptr av packet at the end of the logic helps with this issue 👍

fyi @firesurfer

firesurfer, 2023-10-31:
Thanks for the info.

flynneva, 2023-11-04:
@firesurfer could you test out #290 to see if it fixes your issue? I'm not really sure how to test to confirm this issue is resolved. If you have any ideas lmk, maybe we could add an integration test for it 👍🏼

firesurfer, 2023-11-06:
I will test this hopefully today.

As an idea for an integration test (I didnt take a look at the current tests):

You only need to generate a test stream of images at 30fps with 1920 x 1080 resolution and check if the resulting publishing rate is ~30Hz. 

A simpler to implement intermediate check could be to introduce an update rate counter/timer for the publish calls and check if it averages at 30Hz

firesurfer, 2023-11-06:
@flynneva Just tested #290 

Unfortunately it seems to make it worse. I only achieve 14fps on the raw image topic with this patch. On the compressed topic `ros2 topic hz` shows 25fps. 

For completeness a recap what I did for testing:

1. Adapt the camera config to 1920x1080
2. Launch the camera node
3. Subscribe to the raw image topic with rviz2
4. run ros2 topic hz

firesurfer, 2023-12-19:
Hey @flynneva  did you manage to reproduce this issue by now? (E.g. by providing the test image stream?)

firesurfer, 2024-06-12:
Hey  @flynneva I have another update for you: I realized that even with 720p  (1280 x 720) the frame rate drops the more cameras I add. E.g. at 7 cameras at the moment I only get ~15-20fps. The cameras are distributed to different USB host controllers. 
The system is far from being cpu capped btw. 

So my guess would be that either something in the usb kernel subsystem limits the amount of data we can put through there or there is some kind of locking happening that makes it worse the more cameras we add.

flynneva, 2024-06-12:
@firesurfer interesting. It's possible we are hitting the network buffer limits then since ROS2 still uses them even for local topics/messages.

Can you try increasing the buffers and retesting?

https://stackoverflow.com/questions/62227100/why-isnt-increasing-the-networking-buffer-sizes-reducing-packet-drops

flynneva, 2024-06-16:
@firesurfer im realizing now that this is probably a QoS setting thing 😓  I should've noticed this before. Let me dig into it a bit today and see if we can quickly fix this with adjusting the QoS for the publishers

Woojin-Crive, 2024-10-03:
Using SensorDataQoS has been somewhat helpful, but the image_raw topic still does not reach 30Hz when using the Logitech C920 at a resolution of 1920x1080 on an Intel NUC. (The camera performs smoothly when using the default Ubuntu camera app, Cheese.)

67617liu, 2024-10-18:
Hi, I've also face this issue. I'm using ROS2 Humble, CycloneDDS, and using the camera on my laptop. Whether I use 640 x 480 or 1280 x 720 at 30hz, the output of the `ros2 topic hz` the output is always 10hz. I've also tried switching to SensorQos, but it didn't help.

gaoyuheng123, 2026-08-28:
I reproduced a very similar issue with `usb_cam` 0.8.1 on ROS 2 Humble.

Configuration:

- V4L2 USB camera
- MJPEG input
- `mjpeg2rgb`
- 640x480 at 30 FPS
- `mmap`

The camera and decoding pipeline were able to sustain 30 FPS:

- Direct `v4l2-ctl` capture: approximately 30 FPS
- FFmpeg MJPEG decoding and RGB24 conversion: approximately 30 FPS

However, the ROS topics were significantly slower:

- `/image_raw`: 17.37 Hz using a minimal C++ subscriber
- `/camera_info`: 15.30 Hz
- Median interval between the original V4L2 image timestamps: 67.96 ms

The C++ subscriber only counted messages and performed no image processing. This
also rules out `ros2 topic hz` overhead as the cause.

The current node uses two wall timers with the same requested frame period:

1. `update()` waits for the V4L2 frame and performs conversion.
2. `publish()` publishes the shared image and CameraInfo messages.

When running under the generated single-threaded component executable, the
blocking acquisition/conversion callback delays the separate publish timer.
The shared image message can be updated again before the publish callback runs,
so approximately every second captured frame is skipped.

I tested the following minimal change:

- Remove the separate `m_publish_timer`.
- Call the existing `publish()` function immediately after a successful capture
  inside `update()`.

Conceptually:

```cpp
void UsbCamNode::update()
{
  if (!m_camera->is_capturing()) {
    return;
  }

  const bool success =
    (m_parameters.pixel_format_name == "mjpeg") ?
    take_and_send_image_mjpeg() :
    take_and_send_image();

  if (!success) {
    RCLCPP_WARN_ONCE(
      this->get_logger(),
      "USB camera did not respond in time.");
    return;
  }

  publish();
}

TanayK07, 2026-10-02:
Worth separating two things here: does usb_cam actually publish 30 frames/s, or can't the subscribers keep up? ros2 topic hz on raw 1080p is expensive in itself, since a Python subscriber receives every multi-MB frame (I measured hz at 7.1% of a core on just a 100 KB topic at 50 Hz). RViz on the same topic adds another full copy. To count publishes inside the driver without subscribing, I wrote ros2_pulse (LD_PRELOAD over the tracetools hooks, works with Humble binaries, https://github.com/TanayK07/ros2_pulse). Run pulse-top and compare the publish rate on /image_raw with what your subscriber receives. If publish is already around 17, the loss is in capture or conversion, not DDS. Caveat: it only sees processes launched with it preloaded.