ros-drivers/usb_cam #296: [bug] After starting the camera, the image is blue screen within 20s, and the image is displayed normally after 20s.

zymouse, 2023-11-13:
## work out in advance
- [x] branch: ros2
- [x] commit:  c052be03fe0bc067fdf0941394a62dcc81925ae8
## rendering
[image]

flynneva, 2023-11-17:
@zymouse could you give more info here like what format you're using, frame rate, etc. and then maybe some commands to reproduce?

zymouse, 2023-11-20:
@flynneva 
"video_device": "/dev/video4",
        "framerate": 30.0,
        "io_method": "mmap",
        "frame_id": "camera_front_link",
        "pixel_format": "uyvy2rgb",
        "image_width": 1920 ,
        "image_height": 1080,
        "camera_name": "camera_front",
        "camera_info_url": "file://${ROS_HOME}/../pix/parameter/sensor_kit/robobus_sensor_kit_description/intrinsic_parameters/,.yaml",
        # reusing same camera intrinsics only for demo, should really be unique for camera2"
        "brightness": -1,
        "contrast": -1,
        "saturation": -1,
        "sharpness": -1,
        "gain": -1,
        "auto_white_balance": True,
        "white_balance": 4000,
        "autoexposure": True,
        "exposure": 100,
        "autofocus": False,
        "focus": -1,

flynneva, 2023-12-16:
@zymouse could you try again with the latest usb_cam ros2 branch? It should help you choose a properly supported pixel format for the camera you have and might fix your issue here

flynneva, 2023-12-16:
@zymouse also if you want to visualize the images in rqt I recommend using one of the '2rgb' pixel formats so the published format is compatible with rqt

zymouse, 2024-01-09:
@flynneva 
Thank you very much for your answer: the pixel format I chose is correct, and after starting the usb_cam node, I need to wait for about 5s before the image is converted from a blue screen to a normal image result. This doesn't affect the usage, this issue can be closed now

zymouse, 2024-08-23:
updates：
commit：501af3d428513b2fd520b7a087ba5774b4d6181d
branch:  ros2
```
        "video_device": "/dev/video0",
        "framerate": 30.0,
        "io_method": "mmap",
        "frame_id": "test_camera",
        "pixel_format": "uyvy2rgb",  
        "image_width": 1920,
        "image_height": 1080,
        "camera_name": "test_camera",
        # reusing same camera intrinsics only for demo, should really be unique for camera2"
        "camera_info_url": "file:///test/camera_info_3mm.yaml",
        "brightness": -1,
        "contrast": -1,
        "saturation": -1,
        "sharpness": -1,
        "gain": -1,
        "auto_white_balance": True,
        "white_balance": 4000,
        "autoexposure": True,
        "exposure": 100,
        "autofocus": False,
        "focus": -1,
```
Startup will be blue screen for a period of time, during the blue screen, cpu usage is 100
After the blue screen ends, the image returns to normal, and the cpu usage returns to normal.

[image]
[image]
[image]

flynneva, 2025-02-08:
I cant reproduce this, so closing until we have reproducible instructions

closed 2025-02-08 by flynneva