ros-drivers/usb_cam #299: usb_cam works on galactic but does not work on humble

lmendyk, 2023-11-16:
I’m in a process of migration with my turtlebot3 from galactic to humble ros2 with usb_cam used for raspberry pi 4 camera.  The same code (in respect to usb_cam and image transport) works on galactic.  But when I try to use it inside docker container (started with –prevailed option), the usb_camere does not work.
This means that in the logs I get:

```
[usb_cam_node_exe-4] [INFO] [1700139031.778458873] [camera.usb_cam]: camera_name value: camera
[usb_cam_node_exe-4] [WARN] [1700139031.778748632] [camera.usb_cam]: framerate: 10.000000
[usb_cam_node_exe-4] [INFO] [1700139031.801688143] [camera.usb_cam]: camera calibration URL: package://turtlebot3_manipulation_bringup/camera_info/turtlebot3_rpicamera.yaml
[usb_cam_node_exe-4] [INFO] [1700139031.803539599] [camera.usb_cam]: Starting 'camera' (/dev/video0) at 1920x1080 via mmap (yuyv) at 10 FPS
[usb_cam_node_exe-4] [INFO] [1700139031.820321373] [camera.usb_cam]: This devices supproted formats:

```

And no formats are printed while when running on galactic ROS2 – I get long list of formats

Then I get following logs:

```
[usb_cam_node_exe-4] [INFO] [1700139031.820714983] [camera.usb_cam]: Setting 'brightness' to 50
[usb_cam_node_exe-4] unknown control 'white_balance_temperature_auto'
[usb_cam_node_exe-4] 
[usb_cam_node_exe-4] [INFO] [1700139031.845867690] [camera.usb_cam]: Setting 'white_balance_temperature_auto' to 1
[usb_cam_node_exe-4] [INFO] [1700139031.846006245] [camera.usb_cam]: Setting 'exposure_auto' to 3
[usb_cam_node_exe-4] unknown control 'exposure_auto'
[usb_cam_node_exe-4] 
[usb_cam_node_exe-4] [INFO] [1700139031.859236161] [camera.usb_cam]: Setting 'focus_auto' to 0
[usb_cam_node_exe-4] unknown control 'focus_auto'
[usb_cam_node_exe-4] 
[usb_cam_node_exe-4] [INFO] [1700139032.271723424] [camera.usb_cam]: Timer triggering every 100 ms

```

when trying to display the images in rqt I get

`ImageView.callback_image() while trying to convert image from 'yuv422_yuy2'  to 'rgb8' an exception was thrown (Image is wrongly formated ...) `

flynneva, 2023-11-17:
@lmendyk galactic has been deprecated for awhile now and so it hasn't been getting the recent updates.

Check the read me about selecting the right formats, that's changed recently and could explain your issue here.

lmendyk, 2023-11-17:
@flynneva  - the problem is on humble (on galactic it works!).  I read the readme and don't know what is the issue :(

lmendyk, 2023-11-22:
Ok, I have found it (not in the documentation) that I need to add 
` pixel_format: "yuyv2rgb"`

Still no formats are printed, but at least the camera works so I can see it in rviz and rqt

closed 2023-11-22 by lmendyk

flynneva, 2023-11-27:
@lmendyk this is documented in the README in [this section](https://github.com/ros-drivers/usb_cam/tree/ros2#driver-supported-formats). Glad you got it working!

lmendyk, 2023-11-27:
Sorry - I must have been tricked by the "[default](https://github.com/ros-drivers/usb_cam/tree/develop)" branch  of README which directed me by

> "For full documentation, see [the ROS wiki](http://ros.org/wiki/usb_cam)."

to the obsolete documentation.