ros-drivers/usb_cam #307: CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:463 (message):   A required package was not found

Bhaveshmeghwal21, 2023-12-29:
I am using ROS Noetic 
after cloning this package in my workspace ( ros_ws_cam/src/) it shows a cmake error, the message is 
__CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:463 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:643 (_pkg_check_modules_internal)
  usb_cam/CMakeLists.txt:10 (pkg_check_modules)__
can anyone solve it?

Li-yinqiao, 2024-03-05:
> 在我的工作区（ros_ws_cam/src/）中克隆此包后，我正在使用ROS Noetic，它显示cmake错误，消息为_/usr/share/cmake-3.16/Modules/FindPkgConfig.cmake：463的_CMake错误（消息）：找不到所需的软件包 调用堆栈（最近一次调用优先）：/usr/share/cmake-3.16/Modules/FindPkgConfig.cmake：643 （_pkg_check_modules_internal） usb_cam/CMakeLists.txt：10 （pkg_check_modules）__谁能解决它？

Excuse me, have you solved this problem yet?

twdragon, 2024-03-05:
It seems a problem with ROS installation or sourcing. Can you share a full log?

zhoushiguang7, 2024-10-28:
> I am using ROS Noetic after cloning this package in my workspace ( ros_ws_cam/src/) it shows a cmake error, the message is __CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:463 [...]

I also encountered this problem. It should be because of "No package 'libv4l2' found".
And I solved this error by running sudo apt-get install libv4l-dev in the terminal.

flynneva, 2025-02-08:
Closing due to inactivity

closed 2025-02-08 by flynneva