ros-drivers/usb_cam #308: Node exits because of pixel formatting issues？

zymouse, 2024-01-08:
- [x] branch: ros2
- [x] commit: f5d3e5be625ccae17019d85a27a9c1304819d00a

@flynneva 
`v4l2-ctl --device=/dev/video0 --all`
[image]
Because of a camera firmware error, the pixel format is actually `uyvy`.

When I use the correct pixel format `uyvy`, the usb_cam node exits incorrectly
[image]

flynneva, 2024-01-09:
@zymouse so are you asking for an option to "skip" the check and continue on with the selected format?

I guess that can be added.

zymouse, 2024-01-09:
@flynneva 
Thank you very much for your answer: the option to "skip" the check and continue with the selected format, can you tell me how to change it in the code?

flynneva, 2024-01-12:
@zymouse we'd have to add it still, but essentially it would include adding a new parameter to the yaml file, declaring and setting the parameter in the cpp code and then checking that bool within the new format validation logic we have now.

I can add this maybe over the weekend if I find some time, if not this weekend next week 👍

zymouse, 2024-01-25:
@flynneva Good afternoon, did this issue, get resolved? Thank you very much.

flynneva, 2024-01-25:
@zymouse hey hey! Not yet but I'll take a look this weekend. Sorry it's been a busy start to the new year

flynneva, 2024-01-28:
@zymouse can you try out this branch: https://github.com/ros-drivers/usb_cam/pull/317

and set the new parameter `skip_format_check` to `true` for your hardware and sees if it fixes your issue?

flynneva, 2024-01-31:
@zymouse so to support your buggy camera you'll need the branch from #317 and also to modify [this line](https://github.com/ros-drivers/usb_cam/blob/ros2/src/usb_cam.cpp#L435) to:

```diff
- m_image.v4l2_fmt.fmt.pix.pixelformat = m_image.pixel_format->v4l2();
+m_image.v4l2_fmt.fmt.pix.pixelformat = V4L2_PIX_FMT_YUYV;
```

Try that out and let me know how it goes

zymouse, 2024-02-05:
Okay. Thank you very much.

flynneva, 2024-03-16:
Closing as complete / wont fix on the main line here since it is such a unique problem

closed 2024-03-16 by flynneva