ros-drivers/usb_cam #310: Udev symlinks for webcams are not accepted anymore

firesurfer, 2024-01-15:
Hi @flynneva 
I ran into a new issue this morning. 

On our system we use symlink in order to create `/dev/...` symlinks for each camera depending on the usb port they are connected to. This worked fine until recently (unfortunately I am not sure which update broke that).

I now get the error message from the usb_cam node (exemplary for one camera - we have four) :

```
[usb_cam_node_exe-3] [ERROR] [1705303215.150563105] [camera_3.camera_3_node]: Device specified is not available or is not a vaild V4L2 device: `/dev/camera00`
[usb_cam_node_exe-3] [INFO] [1705303215.150648987] [camera_3.camera_3_node]: Available V4L2 devices are:
[usb_cam_node_exe-3] [INFO] [1705303215.150666700] [camera_3.camera_3_node]:     /dev/video0
```

But using the symlink via vlc or with v4l2-ctl works:

```
v4l2-ctl -D -d /dev/camera00
Driver Info:
	Driver name      : uvcvideo
	Card type        : HD Pro Webcam C920
	Bus info         : usb-0000:02:00.0-2
	Driver version   : 6.2.16
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : HD Pro Webcam C920
	Serial           :
	Bus info         : usb-0000:02:00.0-2
	Media version    : 6.2.16
	Hardware revision: 0x0000000c (12)
	Driver version   : 6.2.16
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : HD Pro Webcam C920
	Function         : V4L2 I/O
	Flags            : default
	Pad 0x01000007   : 0: Sink
	  Link 0x0200001c: from remote pad 0x100000a of entity 'Processing 3' (Video Pixel Formatter): Data, Enabled, Immutable
```

The corresponding udev entry:

```
SUBSYSTEM=="video4linux",KERNELS=="1-2:1.0",SUBSYSTEMS=="usb",DRIVERS=="uvcvideo",ATTR{index}=="0",SYMLINK+="camera00"
```

Usb cam version: 0.8.0

firesurfer, 2024-01-15:
@flynneva  I found the issue: https://github.com/ros-drivers/usb_cam/commit/af45135da45cf93dfa663f6bb022e5858a028b78

I suggest to check if the passed name is a symlink and if yes to resolve it:

BEG-Katzenberger, 2024-01-18:
It's definitely the update from 0.7.0 to 0.8.0.
Did a cross check with build from sources:

### Video devices and symlink
```bash
~/tmp/cam_test_ws$ ls -la /dev/video*
crw-rw----+ 1 root video 81, 0 Jan 18 11:09 /dev/video0
crw-rw----+ 1 root video 81, 1 Jan 18 11:09 /dev/video1
crw-rw----+ 1 root video 81, 2 Jan 18 11:09 /dev/video2
crw-rw----+ 1 root video 81, 3 Jan 18 11:09 /dev/video3
crw-rw----+ 1 root video 81, 4 Jan 18 11:10 /dev/video4
crw-rw----+ 1 root video 81, 5 Jan 18 11:10 /dev/video5

~/tmp/cam_test_ws$ ls -la /dev/micro*
lrwxrwxrwx 1 root root 6 Jan 18 11:10 /dev/microsoft_webcam -> video4
```

### driver launch with 0.8.0
```bash
~/tmp/cam_test_ws/src/usb_cam$ git status
HEAD detached at 0.8.0
...
[INFO] [launch]: All log files can be found below ...
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [usb_cam_node_exe-1]: process started with pid [331620]
[INFO] [camera_diagnostics-2]: process started with pid [331622]
[usb_cam_node_exe-1] [INFO] [1705577554.536271979] [web_cam]: camera_name value: web_camera
[usb_cam_node_exe-1] [WARN] [1705577554.536332710] [web_cam]: framerate: 20.000000
[usb_cam_node_exe-1] [INFO] [1705577554.537362010] [web_cam]: camera calibration URL: package://camera_launch/config/microsoft_lifecam_studio_info.yaml
[usb_cam_node_exe-1] [ERROR] [1705577554.780940459] [web_cam]: Device specified is not available or is not a vaild V4L2 device: `/dev/microsoft_webcam`
[usb_cam_node_exe-1] [INFO] [1705577554.781010102] [web_cam]: Available V4L2 devices are:
[usb_cam_node_exe-1] [INFO] [1705577554.781017259] [web_cam]:     /dev/video0
[usb_cam_node_exe-1] [INFO] [1705577554.781022664] [web_cam]:         HP HD Camera: HP HD Camera
[usb_cam_node_exe-1] [INFO] [1705577554.781028815] [web_cam]:     /dev/video1
[usb_cam_node_exe-1] [INFO] [1705577554.781032047] [web_cam]:         HP HD Camera: HP HD Camera
[usb_cam_node_exe-1] [INFO] [1705577554.781035310] [web_cam]:     /dev/video2
[usb_cam_node_exe-1] [INFO] [1705577554.781038796] [web_cam]:         HP HD Camera: HP IR Camera
[usb_cam_node_exe-1] [INFO] [1705577554.781042089] [web_cam]:     /dev/video3
[usb_cam_node_exe-1] [INFO] [1705577554.781045336] [web_cam]:         HP HD Camera: HP IR Camera
[usb_cam_node_exe-1] [INFO] [1705577554.781048537] [web_cam]:     /dev/video4
[usb_cam_node_exe-1] [INFO] [1705577554.781051641] [web_cam]:         Microsoft® LifeCam Studio(TM):
[usb_cam_node_exe-1] [INFO] [1705577554.781054840] [web_cam]:     /dev/video5
[usb_cam_node_exe-1] [INFO] [1705577554.781057976] [web_cam]:         Microsoft® LifeCam Studio(TM):
[usb_cam_node_exe-1] [WARN] [1705577554.781605666] [web_cam]: Shutting down
[INFO] [usb_cam_node_exe-1]: process has finished cleanly [pid 331620]
```

### driver launch with 0.7.0
```bash
git status
HEAD detached at 0.7.0
nothing to commit, working tree clean
...
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [usb_cam_node_exe-1]: process started with pid [354282]
[INFO] [camera_diagnostics-2]: process started with pid [354284]
[usb_cam_node_exe-1] [INFO] [1705578667.086366283] [web_cam]: camera_name value: web_camera
[usb_cam_node_exe-1] [WARN] [1705578667.086467642] [web_cam]: framerate: 20.000000
[usb_cam_node_exe-1] [INFO] [1705578667.087954263] [web_cam]: camera calibration URL: package://camera_launch/config/microsoft_lifecam_studio_info.yaml
[usb_cam_node_exe-1] [INFO] [1705578667.088214346] [web_cam]: Starting 'web_camera' (/dev/microsoft_webcam) at 800x600 via mmap (mjpeg2rgb) at 20 FPS
[usb_cam_node_exe-1] [INFO] [1705578667.218268894] [web_cam]: This devices supproted formats:
[usb_cam_node_exe-1] [INFO] [1705578667.218771104] [web_cam]: 	YUYV 4:2:2: 640 x 480 (30 Hz)
[usb_cam_node_exe-1] [INFO] [1705578667.218815090] [web_cam]: 	YUYV 4:2:2: 640 x 480 (20 Hz)
[usb_cam_node_exe-1] [INFO] [1705578667.218827984] [web_cam]: 	YUYV 4:2:2: 640 x 480 (15 Hz)
[usb_cam_node_exe-1] [INFO] [1705578667.218838915] [web_cam]: 	YUYV 4:2:2: 640 x 480 (10 Hz)
[usb_cam_node_exe-1] [INFO] [1705578667.218849777] [web_cam]: 	YUYV 4:2:2: 640 x 480 (7 Hz)
...
```

firesurfer, 2024-01-22:
Given that #311 is merged we can close this.

closed 2024-01-22 by firesurfer