ros-drivers/usb_cam #321: package.xml specifies catkin

iplayfast, 2024-03-04:
I thought Ros2 used colcon
I'm having problems building this from source.

iplayfast, 2024-03-04:
Oh, I need to checkout the ros2 branch. I know it says in the read me, but I took it that it was a link to a different repo. 
Can you update the readme to be a bit more clear. Maybe use hand holding instructions to save poor souls like me some time.

flynneva, 2024-03-06:
@iplayfast the read me is tied to each branch, so we'd have to update the readme on every branch to each other branch.

Instead I'd rather just spend time reducing the number of branches we need. Right now there is a prototype on the ROS 2 branch that uses the same cpp base library for both ros 1 and ros 2.

closed 2024-03-06 by flynneva