ros-drivers/usb_cam #332: Error in package.xml

yuntianli91, 2024-05-11:
Hi, @flynneva, thanks to your great job.

I realized you have add support for ROS1 since version 0.8.0, however I think there is a mistyping in the `<export>` label of `package.xml`, which are:

```xml
...
<export>
    <build_type condition="$ROS_VERSION == 1">ament_cmake</build_type>
    <build_type condition="$ROS_VERSION == 2">ament_cmake</build_type>
  </export>
...
```
however the `buile_type` should be `catkin` for ROS1. I changed this label and it works for ROS1 now.

yuntianli91, 2024-05-11:
Also there are some error in the function iterface (such as `m_camera.start()`, e.t.c) in the executable `src/ros1/usb_cam_node.cpp`, I'm will check how to fix it ...

yuntianli91, 2024-05-11:
> Also there are some error in the function iterface (such as `m_camera.start()`, e.t.c) in the executable `src/ros1/usb_cam_node.cpp`, I'm will check how to fix it ...

The problem is the `take_and_send_image()` function, I past my version below:
```cpp
 bool take_and_send_image()
  {
    // grab the new image
    auto new_image = m_camera.get_image();

    // fill in the image message
    m_image.header.stamp.sec = m_camera.get_image_timestamp().tv_sec;
    m_image.header.stamp.nsec = m_camera.get_image_timestamp().tv_nsec;

    // Only resize if required
    if (m_image.data.size() != static_cast<size_t>(m_camera.get_image_step() * m_camera.get_image_height())) {
      m_image.width = m_camera.get_image_width();
      m_image.height = m_camera.get_image_height();
      m_image.encoding = m_camera.get_pixel_format()->ros();
      m_image.step = m_camera.get_image_step();
      m_image.data.resize(m_camera.get_image_step() * m_camera.get_image_height());
    }

    // Fill in image data
    memcpy(&m_image.data[0], new_image, m_image.data.size());

    // grab the camera info
    sensor_msgs::CameraInfoPtr ci(new sensor_msgs::CameraInfo(m_camera_info->getCameraInfo()));
    ci->header.frame_id = m_image.header.frame_id;
    ci->header.stamp = m_image.header.stamp;

    // publish the image
    m_image_pub.publish(m_image, *ci);

    return true;
  }
```

flynneva, 2024-05-11:
@yuntianli91 yes the plan was to merge together the divergent ros1 and ROS2 branches eventually so they can use the same underlying base USB cam library, just I never had time to finish it up 

Thanks for the effort / debugging here, is it ok if I open a PR to fix this or did you want to?

yuntianli91, 2024-05-11:
> @yuntianli91 yes the plan was to merge together the divergent ros1 and ROS2 branches eventually so they can use the same underlying base USB cam library, just I never had time to finish it up
> 
> Thanks for the effort / debugging here, is it ok if I open a PR to fix this or did you want to?

Well, I think it's better to open a PR to fix it as you're much more familier with the source code, I'm quite looking forward to your future release. 
Thanks in advance ^_^ !

closed 2024-05-16 by yuntianli91