ros-drivers/usb_cam #350: Support for Y16 Pixel Format

rancyberbee, 2024-11-14:
Hi
My Arducam AR0234 support the following formats:

ubuntu@ubuntu:~/ros2_ws$ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'Y16 ' (16-bit Greyscale)
                Size: Discrete 1920x1200
                Size: Discrete 1920x1080
                Size: Discrete 1280x720
        [1]: 'BA10' (10-bit Bayer GRGR/BGBG)
                Size: Discrete 1920x1200
                Size: Discrete 1920x1080
                Size: Discrete 1280x720
                
From my knowledge it is pretty similar to mono16 but with some additional metadata or encoding. Is there a support for "Y16" pixel format for your driver for ROS2?

Thanks,
Ran

flynneva, 2024-11-27:
@rancyberbee not as of today but I'd be happy to review a PR if you want to take a crack at adding support for that format!

You can take what is in the 'include/usb_cam/formats/mono16.hpp' file as inspiration.

If you open a PR I could comment and guide you through adding it yourself

flynneva, 2025-02-08:
docs for Y16: https://www.kernel.org/doc/html/v4.17/media/uapi/v4l/pixfmt-y16.html?highlight=y16

flynneva, 2025-02-08:
@rancyberbee oh - Y16 is already supported actually, we just call it `MONO16`:  https://github.com/ros-drivers/usb_cam/blob/main/include/usb_cam/formats/mono.hpp#L67

flynneva, 2025-02-08:
Closing as completed

closed 2025-02-08 by flynneva