ros-drivers/usb_cam #351: Issue with catkin_make and USB Camera Integration in ROS Noetic

cxx-ovo, 2024-11-16:
Hello ROS Community,

I'm currently working with ROS Noetic and trying to integrate a USB camera into my project using the usb_cam package. However, I'm encountering several issues during the build process, and I need some help resolving them.

Problem Details:
Workspace Build Error: When I try to build my workspace using catkin_make, I get the following error:
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkin_workspace.cmake:100 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the 'catkin_make_isolated' command instead.
It seems like the usb_cam package is not properly recognized as a Catkin package, and the workspace contains non-Catkin packages, causing this error.

Warning about image_transport_plugins: I also get a warning when running the build:
WARNING: package "usb_cam" should not depend on metapackage "image_transport_plugins" but on its packages instead
This warning suggests that usb_cam is incorrectly depending on a metapackage instead of the individual components.

Steps I've Tried:
I tried running catkin_make_isolated, but the issue persists.
I attempted to modify the CMakeLists.txt file in the usb_cam package to remove the metapackage dependency (image_transport_plugins), but it hasn't solved the problem yet.
Request for Help:
Can anyone help me understand why the workspace contains non-Catkin packages, and how to resolve the build error?
Is there a correct way to configure the usb_cam package to avoid the image_transport_plugins dependency?
Are there any other solutions to build the workspace correctly and get the USB camera working?
I would greatly appreciate any advice or solutions!

knorth55, 2024-12-04:
I recommend to use `catkin-tools` instead of `catkin_make`.
also for the dependency installation, please use `rosdep`.

mxhhhhhhh, 2024-12-04:
i also met this problem. Have you found a way to solve it? i'm appreciate it if you  can tell me.

knorth55, 2024-12-04:
did you source ros `/opt/ros/noetic/setup.bash` before building the package?

knorth55, 2024-12-04:
Ah, now I understood. please use `develop` branch for `ros1`.
the default branch is for `ros2`.

closed 2024-12-04 by knorth55