ros-drivers/usb_cam #352: humble launch fails

anfederman, 2024-11-16:
ros2 launch usb_cam camera.launch.py
[INFO] [launch]: All log files can be found below /home/ubiqlap/.ros/log/2024-11-16-08-31-53-762719-ubiqlap-X540LA-5691
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught multiple exceptions when trying to load file of format [py]:
 - PydanticUserError: If you use `@root_validator` with pre=False (the default) you MUST specify `skip_on_failure=True`. Note that `@root_validator` is deprecated and should be replaced with `@model_validator`.

For further information visit https://errors.pydantic.dev/2.9/u/root-validator-pre-skip
 - InvalidFrontendLaunchFileError: The launch file may have a syntax error, or its format is unknown

flynneva, 2024-11-27:
@anfederman thanks for reporting the error! Looks like we need to update the launch file to the latest pydantic syntax.

I won't have time to tackle this anytime soon unfortunately, but would happily review a PR

flynneva, 2025-02-08:
@anfederman so I believe this is only an issue if you use `pydantic>=2.9` it looks like.

For Humble - using `rosdep` to install pydantic would be like calling `sudo apt-get install python3-pydantic`....which isnt > 2.9.

Even for Rolling on Ubuntu 24.04 (what I have locally) it will still install an older version of pydantic: https://launchpad.net/ubuntu/+source/pydantic

So not sure I want to make this fix just yet, but will probably have to make it once Rolling switches to a new Ubuntu distro.